KrisKasprzak/EBYTE #1: Problem with e32 module communication with ttgo esp32 lora

Alltimetech, 2019-03-29:
hello,  I am trying to communicate this type of ebyte e32 modules with ttgo lora esp 32, but it has not been possible to change the addresses, the frequency set them to 915 mhz and I have not yet been able to communicate them between them ..... I hope you can help me thanks

KrisKasprzak, 2019-03-29:
Are you trying to communicate between two different modules or the same E32 for send and receive? Were you able to get the examples running?

Alltimetech, 2019-03-29:
Im using  E32 as transmitter and TTGO LoRa esp32 as receiver.
I tried to do it, but I cannot get any packets even tho I send a broadcast packet. I was writing to E32 support and they said following:

"Because our module is flashed by our own firmware, so we cannot make sure the compatibility with others, pls understand."

So, I don't understand if it's one LoRa standard how it can be incompatible?

KrisKasprzak, 2019-03-29:
Not sure I can help. My library is for using and programming and E32 from EBYTE. I have no idea what it will take to program and use the esp32. I suspect it will be vastly different.

KrisKasprzak, 2019-03-29:
Have a look at a library from RadioHead. Maybe that lib has some code for the esp32

Alltimetech, 2019-03-29:
Hi, I have 2 esp32 and communicate very well with each other, but when I try to communicate with the e32 using your library I do not receive or send the packages

KrisKasprzak, 2019-03-29:
Yep, as I said and so did the mfg, EBYTE only works with EBYTE

closed 2020-10-04 by KrisKasprzak