KrisKasprzak/EBYTE #3: FEC configuration

philkfr, 2019-05-09:
Hi,
Thank you for this excellent library.
I do not know how I did it, but I have an E32-868T20D module where Sped = 2 and Optn = 0.
I would like to change these settings (mostly because FEC = 0), but I did not find the method. Can you help me ? Thank you in advance.
Philippe

KrisKasprzak, 2019-05-09:
Can you just use

SetAirDataRate (val); 
or 
SetUARTBaudRate (val); 

Not sure what you are asking for.

Thanks,

Kris Kasprzak
256-348-0953
kris.kasprzak@yahoo.com

philkfr, 2019-05-10:
Thank you for your reply.
Finally, I took inspiration from your library to write a procedure to change the SPED and OPTION bytes manually.
Everything is in order.
Thanks.

closed 2019-05-10 by philkfr