KrisKasprzak/EBYTE #6: PrintParameters() not working with E32-TTL-100

vitorjrodrigues, 2019-06-23:
I'm using an E32-TTL-100 wired exactly as described in the example and didn't modify a single line of code except by commenting the loop content. Meaning it should just print the current parameters of the module.

But somehow, every value I'm getting printed on screen is 0.
I've tested this on two modules and they both are able to send data to each other, but unable to shown and/or set any configuration through this code.

KrisKasprzak, 2019-08-29:
> Hmm, I'm having the same problem except they are not receiving data, only sending. I'm using an E32-915T20D, which I have yet to get to receive successfully from any unit so far.

What MCU are you using? Same MCU for both transceivers?

Thanks,
Kris Kasprzak

KrisKasprzak, 2019-09-19:
I just tested using a NANO and no issues. Can you be more specific as to the MCU. Not much i can do w/o basic info.

account75783, 2019-09-26:
Hi, I get 

 -> ----------------------------------------
 -> Model no.: 0
-> Version  : 0
 -> Features : 0
 ->  
-> Mode (HEX/DEC/BIN): 0/0/0
 -> AddH (HEX/DEC/BIN): 0/0/0
 -> AddL (HEX/DEC/BIN): 0/0/0
 -> Sped (HEX/DEC/BIN): 0/0/0
 -> Chan (HEX/DEC/BIN): 0/0/0
 -> Optn (HEX/DEC/BIN): 0/0/0
 -> Addr (HEX/DEC/BIN): 0/0/0
 ->  
 -> SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
 -> SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
 -> SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
 -> OptionTrans (HEX/DEC/BIN)       : 0/0/0
-> OptionPullup (HEX/DEC/BIN)      : 0/0/0
-> OptionWakeup (HEX/DEC/BIN)      : 0/0/0
 -> OptionFEC (HEX/DEC/BIN)         : 0/0/0
 -> OptionPower (HEX/DEC/BIN)       : 0/0/0
 -> ----------------------------------------

Arduino Leonardo, Pro Micro clone

KrisKasprzak, 2019-09-26:
Very possible it's a timing issue in the library, waiting for the digitalPins to change state. In the EBYTE.cpp file around line 261 and 282 there is a delay call SmartDelay(40);
This delay lets the digitalPins react and may be too short, so change it to 
SmartDelay(1000);
Way too long but lets start there.

If still no go, try adding a SmartDelay(10); to line 574, this will give the module time to write it's parameters

Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

KrisKasprzak, 2020-12-06:
I just Uploaded a version that may have fixed this problem. I can't reproduce it but code additions I made clear buffers and such. Can you download the latest in see if it works for you?

closed 2020-12-06 by KrisKasprzak