KrisKasprzak/EBYTE #9: E32-868T20D

alper1059, 2019-09-27:
Hi,
I need some help I can get the code running.
I want to program and address with 1 coordinator 5 slave.
desired address in the communication with the lora on the arduino on led i want to burn. or I want to extinguish.
How can I do it.Thank you.

===Coordinator===
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(11);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();
====Slave===  1
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(1);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();

====Slave===  2
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(2);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();

====Slave===  3
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(3);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();

====Slave===  4
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(4);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();

====Slave===  5
Transceiver.init();
 Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(5);
  Transceiver.SaveParameters(PERMANENT);

  Transceiver.PrintParameters();

KrisKasprzak, 2019-09-27:
Your recievers will never get any data with the code below. Channels MUST all be the same.
For your coordinator do thisTransceiver.init();
Transceiver.SetAirDataRate(ADR_8K); // must be the same on all units
Transceiver.SetAddressH(255);            // address must be 0xFFFF (H=FF, L=FF) or 0x0000 for broadcast mode
Transceiver.SetAddressL(255);
Transceiver.SetChannel(11);                // must be the same on all units
Transceiver.SaveParameters(PERMANENT);Transceiver.SendStruct(&MyData, sizeof(MyData));    // send what ever data you need

====Slave=== 1 to n
Transceiver.init();

Transceiver.SetAirDataRate(ADR_8K);// must be the same on all units

Transceiver.SetAddressH(0);                // for slave the address doesn't matter

Transceiver.SetAddressL(0);                // for slave the address doesn't matterTransceiver.SetChannel(11);                // must be the same on all unitsTransceiver.SaveParameters(PERMANENT);Transceiver.GetStruct(&MyData, sizeof(MyData));

I've uploaded some data sheets to the github site, have a look as they discuss a bit more on broadcast transmission
However. I have a similar setup where i have 1 base and 4 slaves my base sends data to each base but how I do it is by sending and ID in the data structure, each slave has it's own ID. That way I the base can send a signal to each slave. My data packet has an ID field that is populated by the base, each slaves receives the data packet and looks at the ID if a compare is true, the data packet is intended for that slave for example
Slave 1 ID = 1Slave 2 ID = 2Slave 3 ID = 3Slave 4 ID = 4

struct TheData {
uint16_t ID;uint16_t Open;uint16_t Status:
}

// baseMyData.ID = 0;         // send IDMyData.Open = 1; // send instructionTransceiver.SendStruct(&MyData, sizeof(MyData));  

//ReceiverTransceiver.GetStruct(&MyData, sizeof(MyData));
if (MyData.ID == 1) {// incoming data packet is for meServo.SetPostion(MyData.Open); // or something}

Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

alper1059, 2019-10-03:
Thank you for replying,But I cant work,I understand you.I can not run.
////////////Master/////////
#include <SoftwareSerial.h>
#include "EBYTE.h"

#define PIN_M0 4
#define PIN_M1 5
#define PIN_AX 6

int Chan;
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;
  };
struct TheData {uint16_t ID;uint16_t Open;uint16_t Status:};  // not compilation error

DATA MyData;

SoftwareSerial ESerial(2, 3);

EBYTE Transceiver(&ESerial, 4, 5, 6);

void setup() {

  Serial.begin(9600);

  ESerial.begin(9600);

  Serial.println("Starting Sender");

  Transceiver.init();
  Transceiver.SetMode(MODE_NORMAL);
  Transceiver.Reset();

  Transceiver.SetTransmitPower(OPT_TP20);

  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(255);
  Transceiver.SetAddressL(255);
  Transceiver.SetChannel(6);
  Transceiver.SaveParameters(PERMANENT);
  Transceiver.SendStruct(&MyData, sizeof(MyData));

  Transceiver.PrintParameters();
}

  // SetWORTIming(OPT_WAKEUP250);

void loop() {

  MyData.Count++;
  MyData.Bits = analogRead(A0);
  MyData.Volts = MyData.Bits * ( 5.0 / 1024.0 );

  //Transceiver.SendByte('A');
  IDMyData.Open = 11; 
  Transceiver.SendStruct(&MyData, sizeof(MyData));
  Serial.print("Sending: "); Serial.println(MyData.Count); Serial.print(analogRead(A0));
  delay(1000);
  if (Serial.available()) {
    Chan = Serial.read();

    if (Chan > 47) {
      Serial.println(Chan-48);
      Transceiver.SetChannel(Chan-48);
      Transceiver.SaveParameters(PERMANENT);
      Transceiver.PrintParameters();
    }
  }

}

//////Slave  1//////////
#include <SoftwareSerial.h>
#include "EBYTE.h"
 
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};
int Chan;
DATA MyData;

SoftwareSerial ESerial(2, 3);

EBYTE Transceiver(&ESerial, 4, 5, 6);

void setup() {

  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, INPUT);
  pinMode(13, OUTPUT);

  Serial.begin(9600);

  ESerial.begin(9600);
  Serial.println("Starting Reader");

  Transceiver.init();
  Transceiver.Reset();
  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(6);
  Transceiver.SaveParameters(PERMANENT);
  Transceiver.GetStruct(&MyData, sizeof(MyData));

   Transceiver.PrintParameters();

}

void loop() {

 if (ESerial.available()) 
  { if (MyData.ID == 11){          // not compilation error
    Transceiver.GetStruct(&MyData, sizeof(MyData));

    Serial.print("Count: "); Serial.println(MyData.Count);
    Serial.print("Bits: "); Serial.println(MyData.Bits);
    Serial.print("Volts: "); Serial.println(MyData.Volts);
    digitalWrite(13, HIGH);
    delay(1000);
    digitalWrite(13, LOW);
    delay(1000);
    }
  }
  else {
    if (Serial.available()) {
      Chan = Serial.read();

      if (Chan > 47) {
        Serial.println(Chan - 48);
        Transceiver.SetChannel(Chan - 48);
        Transceiver.SaveParameters(PERMANENT);
        Transceiver.PrintParameters();
      }
    }
    delay(1000);
    Serial.println("Searching: ");
  }
}

//////Slave  2//////////
#include <SoftwareSerial.h>
#include "EBYTE.h"
 
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};
int Chan;
DATA MyData;

SoftwareSerial ESerial(2, 3);

EBYTE Transceiver(&ESerial, 4, 5, 6);

void setup() {

  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, INPUT);
  pinMode(13, OUTPUT);

  Serial.begin(9600);

  ESerial.begin(9600);
  Serial.println("Starting Reader");

  Transceiver.init();
  Transceiver.Reset();
  Transceiver.SetAirDataRate(ADR_8K);
  Transceiver.SetAddressH(0);
  Transceiver.SetAddressL(0);
  Transceiver.SetChannel(6);
  Transceiver.SaveParameters(PERMANENT);
  Transceiver.GetStruct(&MyData, sizeof(MyData));

   Transceiver.PrintParameters();

}

void loop() {

 if (ESerial.available()) 
  { if (MyData.ID == 12){        // not compilation error
    Transceiver.GetStruct(&MyData, sizeof(MyData));

    Serial.print("Count: "); Serial.println(MyData.Count);
    Serial.print("Bits: "); Serial.println(MyData.Bits);
    Serial.print("Volts: "); Serial.println(MyData.Volts);
    digitalWrite(13, HIGH);
    delay(1000);
    digitalWrite(13, LOW);
    delay(1000);
    }
  }
  else {
    if (Serial.available()) {
      Chan = Serial.read();

      if (Chan > 47) {
        Serial.println(Chan - 48);
        Transceiver.SetChannel(Chan - 48);
        Transceiver.SaveParameters(PERMANENT);
        Transceiver.PrintParameters();
      }
    }
    delay(1000);
    Serial.println("Searching: ");
  }
}

------Thank you very much for replying------------

KrisKasprzak, 2019-10-03:
see below....
I recommend you spend some time reading the data sheet, and brushing up on programming in C.
there are 2 huge mistakes with this line of code
IDMyData.Open = 11;

 
 

// what is this line for?struct TheData {uint16_t ID;uint16_t Open;uint16_t Status:}; // not compilation error

// why are you calling this line?Transceiver.SetMode(MODE_NORMAL);
// not implemented don't call itTransceiver.Reset();

// why are you sending data that is not yet populated? Why send data in the setup?Transceiver.SendStruct(&MyData, sizeof(MyData));

// what is IDMyData? never declared so I expect and error here
// i have no idea what you are trying to do here// this code is a mess// is master sending and receiving? If so received data is a struct
// what do you expect to read here? why do you think the first byte is the channel--you're transmitting the struct
not sure why you want to change channel, if all channels are not the same, transmission will fail

alper1059, 2019-10-03:
Thank you very much for replying.What am I doing wrong.

Starting Reader
----------------------------------------
Model no.: 45
Version  : C
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 3/3/11
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 0/0/0
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
SpeedAirDataRate (HEX/DEC/BIN)  : 3/3/11
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 0/0/0
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 0/0/0
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
Count: 50331840
Bits: 6
Volts: 0.00
Searching: 
Searching: 
Searching: 
Searching: 

Starting Sender
----------------------------------------
Model no.: 45
Version  : C
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1B/27/11011
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 3/3/11
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
Sending: 1
265
Sending: 2
265
Sending: 3
264

HelmuthBecker, 2020-02-29:
hi, 

what is AddressH and AddressL?

Thanks for help.

KrisKasprzak, 2020-03-02:
Per the datasheet ADDH and ADDL is the address byte of the module for communication with other modules in the same channel
In fixed transmission mode the module can communicate with other modules in different channels. In broadcast model, (address 0xFFFF) the module can communicate with other modules in the same channel
I've never experimented with changing the address as my use is between 2 modules only.
Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

closed 2020-10-04 by KrisKasprzak