KrisKasprzak/EBYTE #10: E32-91520D with SX1276

JRAlvarez2017, 2019-10-06:
Is it possible to communicate a ES32-91520D with a SX1276 (SPI interface with Arduino)? 
Do we know the LoRa settings in the ES32-91520D?

onipot, 2020-04-08:
same here, any news @JRAlvarez2017 ?

KrisKasprzak, 2020-04-08:
I don't see why you can't communicate back and forth using that configuration. From the data sheet you have to be on the same channel, air data rate, and address (if not using address 0xFFFF or 0x0000.
I have this same transceiver communicating between and Arduino and a Teensy 3.2. My lib has updated example code

onipot, 2020-04-08:
`EBYTE Transceiver(&Serial1, 5, 4, 3);
void setup() {

  Serial.begin(9600);
  Serial1.begin(9600);

  pinMode(5, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(3, INPUT);

  Transceiver.init();

  Transceiver.SetChannel(6); // 862 + 6 = 868 in my case
  Transceiver.SetAddress(0x0000);
  Transceiver.SaveParameters(TEMPORARY);
    Transceiver.PrintParameters();
   

}
`
Then in the loop I check if data is available, but nothing.

KrisKasprzak, 2020-10-04:
note this library is only for UART based devices not the SPI ones EBYTE offers

closed 2020-10-04 by KrisKasprzak

renanqts, 2021-04-04:
@masc-it did you find a solution for this?

masc-it, 2021-04-04:
@renanqts check this other [ebyte library ](https://github.com/xreef/LoRa_E32_Series_Library)

yachal-upson, 2022-01-09:
:Bump:
Also really interested in this. 
I rebooted on a project where I'd bought a number of ES32 modules (really easy implementation among themselves).
However I'm trying to achieve communication between a number of Heltec AB01 Cubecells and an ES32 as a base station. 

Can't for the life of me understand what the settings + address etc needs to be; Ebyte have done so much 'under the hood', that isn't plain SX1276. 

Are the ES32's implementing encryption by default? Will they accept packets/frames without Ebyte proprietary headers etc?

varna9000, 2023-10-02:
I'm also interested if e32 modules can communicate with other non ebyte modules (e.g. plain sx126x/sx127x). Is there any settings besides the appended header of address + channel bytes?

KrisKasprzak, 2023-10-03:
I've never experimented with non CBYTE modules based on the same chip. I'm not sure if this lib would support other mfg's. If you have a module in mind, post a link and i'll look at it's datasheet.