KrisKasprzak/EBYTE #11: M0 & M1 optional?

tqueale, 2019-10-21:
I wish to always use my unit in 'normal' mode (M1 & M0 both LOW). Is there an option to assign -1 to these pins during creation of the EBYTE object, similar to that provided for the Aux pin? Both M0 & M1 would be hard wired to Gnd. Saves 2 I/O lines as well.

tqueale, 2019-10-23:
OK, just had a look through the source code. This is not an option so I will need to modify my copy of the library if I continue down that path.

KrisKasprzak, 2019-10-23:
the latest source code allows -1 for M0 and M1
Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

tqueale, 2019-10-23:
It appears that the changes for M0 & M1 also assume that Aux is also not used as you never set pinmode for Aux if M0 & M1 are set to -1. That should not be the case as Aux is still very valid.
Also, at line 62 "if ((_M0 == -1) & (_M1 == -1)){" did you mean '&&'?
Not trying to be picky, just trying to understand.

tqueale, 2019-11-30:
Should I raise a new issue for the query about the use of a bit-wise and (&) at line 61 where I believe a logical and (&&) was intended?
And another for the missing Aux function when M0 & M1 are set to -1?
It appears you may not have seen my previous comments.

closed 2019-11-30 by tqueale