KrisKasprzak/EBYTE #12: Set Parameters Missing

Camarda, 2019-11-03:
The functions to set the three parameters below are missing:

_OptionTrans
_OptionPullup
_OptionFEC

I added the functions in EBYTE.cpp / EBYTE.h and it worked perfectly for my use, but maybe you should add it to your code.

AB-informatica-service, 2020-02-29:
Can you post the functions that you have write? Thanks

KrisKasprzak, 2020-03-02:
I don't understand what you are asking. 
Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

Camarda, 2020-03-02:
There's my modified files [EBYTE.zip](https://github.com/KrisKasprzak/EBYTE/files/4277496/EBYTE.zip)

If I remember well (not really sure), I simply added this on EBYTE.cpp:

`void EBYTE::SetTrans(uint8_t val) {
	_OptionTrans = val;
	BuildOptionByte();
}`

`void EBYTE::SetPullup(uint8_t val) {
	_OptionPullup = val;
	BuildOptionByte();
}`

`void EBYTE::SetFEC(uint8_t val) {
	_OptionFEC = val;
	BuildOptionByte();
}`

And this to EBYTE.h (in class EBYTE):

`	void SetTrans(uint8_t val);
	void SetPullup(uint8_t val);
	void SetFEC(uint8_t val);`

The variables and significant info were already there, there just wasn't a way to modify them.

KrisKasprzak, 2020-10-04:
these functions and more added

closed 2020-10-04 by KrisKasprzak