KrisKasprzak/EBYTE #13: M0, M1 and Aux logical issues

tqueale, 2019-11-30:
It appears that the changes for M0 & M1 also assume that Aux is also not used as you never set pinmode for Aux if M0 & M1 are set to -1. That should not be the case as Aux is still very valid.
Also, at line 61 "if ((_M0 == -1) & (_M1 == -1)){" did you mean '&&' as I believe a logical AND is required here and not a bit-wise AND?
Not trying to be picky, just trying to understand.

_Originally posted by @tqueale in https://github.com/KrisKasprzak/EBYTE/issues/11#issuecomment-545684237_

tqueale, 2020-04-02:
Kris, Any comment on this? It has been a while.

KrisKasprzak, 2020-04-02:
I should not have made this change. you really need to have m0 and m1 connected to be able to program the unit. I'll be backing that change out

closed 2020-04-02 by KrisKasprzak