KrisKasprzak/EBYTE #14: how to get the RSSI or something comparable for the E32868T30D?

knopserl, 2020-01-26:
Are there any possibility to get RSSI value, something compareable or any means to monitor signal/communication quality/strength for the E32868T30D?
I guess the module itself does not provide that? Does it make sense to contact EBYTE to enhace their firmware for tge module to implement that?
I have found statements and code that indicates, that Semtech SX1276/78 supports RSSI (which is most likely used by the E32 modules)

KrisKasprzak, 2020-01-27:
Great question. i'll contact ebyte to see if they can add support.

Thanks,
Kris

knopserl, 2020-01-27:
@KrisKasprzak thanks a lot. I have a couple of the EBYTE (mostly E32, 868MHz) modules and your lib works very well. I'm using it with ESP32 (Lolin32).  But  I really miss the signal quality to monitor my conection quality, which is important to find out what is the best antenna, the max distanz for a stable connection. I'm wondering, why such an important parameter/value is not supported from the beginning, since most RF chips can provide that, also the Semtech LoRa chips.

KrisKasprzak, 2020-01-27:
Glad the lib works, I contacted EBYTE and they are on holiday for the next few weeks--celebrating Chinese New Year. I'll ping them again if I don't hear back. Hopefully they will implement it. I've need such capability too--just never thought about it.
Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

grandehombre, 2020-02-04:
@KrisKasprzak

Hi Kris,
I've made a few minor mods to the PrintParameter function, so that it shows the following settings in text:
UART rate, Air rate, frequency and power.

Would you mind incorporating this into your library. I haven't used pull-requests before and don't want to mess up anything.

I have 100mW (433T20DT) and 1W (433T20DT) modules. 
I have based these changes on the _apparent_ fact that version 0x48 is the 100mW and 0x49 is the 1W module. Would you know if that is correct? I haven't been able to find any info on this.

Thanks for a great library and have a great 2020!
```
void EBYTE::PrintParameters() {

	if (!_prog){
		Serial.println("M0 and M1 pins set to -1, reading can't be done");
		return;
	}

	ReadParameters();		

	_ParityBit = (_Speed & 0XC0) >> 6;
	_UARTDataRate = (_Speed & 0X38) >> 3;
	_AirDataRate = _Speed & 0X07;

	_OptionTrans = (_Options & 0X80) >> 7;
	_OptionPullup = (_Options & 0X40) >> 6;
	_OptionWakeup = (_Options & 0X38) >> 3;
	_OptionFEC = (_Options & 0X07) >> 2;
	_OptionPower = (_Options & 0X03);

	const long uartDataRate[] = {1200, 2400, 4800, 9600, 19200, 34800, 57600, 115200};
	const long airDataRate[] = {300, 1200, 2400, 4800, 9600, 19200, 19200, 19200};
	
	int transmitPower[4];
	switch (_Version) {
		case 0x48:			// 100mW version
			transmitPower[0] = 20;
			transmitPower[1] = 17;
			transmitPower[2] = 14;
			transmitPower[3] = 10;
			break;
			
		case 0x49:			// 1W version
			transmitPower[0] = 30;
			transmitPower[1] = 27;
			transmitPower[2] = 24;
			transmitPower[3] = 21;
			break;
	}	
	
	char *moduleFrequency;
	switch (_Model) {
		case 0x32:
		moduleFrequency = "433";
		break;
		
		case 0x38:
		moduleFrequency = "470";
		break;
		
		case 0x44:
		moduleFrequency = "915";
		break;
		
		case 0x45:
		moduleFrequency = "868";
		break;
		
		case 0x46:
		moduleFrequency = "170";
		break;
	}
	
	Serial.println("----------------------------------------");
	Serial.print(F("Model no.: "));  Serial.println(_Model, HEX);
	Serial.print(F("Version  : "));  Serial.println(_Version, HEX);
	Serial.print(F("Features : "));  Serial.println(_Features, HEX);
	Serial.print(F("Frequency: "));  Serial.print(moduleFrequency);Serial.println("MHz");
	Serial.println(F(" "));
	Serial.print(F("Mode (HEX/DEC/BIN): "));  Serial.print(_Save, HEX); Serial.print(F("/"));  Serial.print(_Save, DEC); Serial.print(F("/"));  Serial.println(_Save, BIN);
	Serial.print(F("AddH (HEX/DEC/BIN): "));  Serial.print(_AddressHigh, HEX); Serial.print(F("/")); Serial.print(_AddressHigh, DEC); Serial.print(F("/"));  Serial.println(_AddressHigh, BIN);
	Serial.print(F("AddL (HEX/DEC/BIN): "));  Serial.print(_AddressLow, HEX); Serial.print(F("/")); Serial.print(_AddressLow, DEC); Serial.print(F("/"));  Serial.println(_AddressLow, BIN);
	Serial.print(F("Sped (HEX/DEC/BIN): "));  Serial.print(_Speed, HEX); Serial.print(F("/")); Serial.print(_Speed, DEC); Serial.print(F("/"));  Serial.println(_Speed, BIN);
	Serial.print(F("Chan (HEX/DEC/BIN): "));  Serial.print(_Channel, HEX); Serial.print(F("/")); Serial.print(_Channel, DEC); Serial.print(F("/"));  Serial.println(_Channel, BIN);
	Serial.print(F("Optn (HEX/DEC/BIN): "));  Serial.print(_Options, HEX); Serial.print(F("/")); Serial.print(_Options, DEC); Serial.print(F("/"));  Serial.println(_Options, BIN);
	Serial.print(F("Addr (HEX/DEC/BIN): "));  Serial.print(_Address, HEX); Serial.print(F("/")); Serial.print(_Address, DEC); Serial.print(F("/"));  Serial.println(_Address, BIN);
	Serial.println(F(" "));
	Serial.print(F("SpeedParityBit (HEX/DEC/BIN)    : ")); Serial.print(_ParityBit, HEX); Serial.print(F("/"));  Serial.print(_ParityBit, DEC); Serial.print(F("/"));  Serial.println(_ParityBit, BIN);
	Serial.print(F("SpeedUARTDataRate (HEX/DEC/BIN) : ")); Serial.print(_UARTDataRate, HEX); Serial.print(F("/"));  Serial.print(_UARTDataRate, DEC); Serial.print(F("/"));  Serial.print(_UARTDataRate, BIN);Serial.print(" ");Serial.print(uartDataRate[_UARTDataRate]);Serial.println("bpm");
	Serial.print(F("SpeedAirDataRate (HEX/DEC/BIN)  : ")); Serial.print(_AirDataRate, HEX); Serial.print(F("/"));  Serial.print(_AirDataRate, DEC); Serial.print(F("/"));  Serial.print(_AirDataRate, BIN);Serial.print(" ");Serial.print(airDataRate[_AirDataRate]);Serial.println("bpm");
	Serial.print(F("OptionTrans (HEX/DEC/BIN)       : ")); Serial.print(_OptionTrans, HEX); Serial.print(F("/"));  Serial.print(_OptionTrans, DEC); Serial.print(F("/"));  Serial.println(_OptionTrans, BIN);
	Serial.print(F("OptionPullup (HEX/DEC/BIN)      : ")); Serial.print(_OptionPullup, HEX); Serial.print(F("/"));  Serial.print(_OptionPullup, DEC); Serial.print(F("/"));  Serial.println(_OptionPullup, BIN);
	Serial.print(F("OptionWakeup (HEX/DEC/BIN)      : ")); Serial.print(_OptionWakeup, HEX); Serial.print(F("/"));  Serial.print(_OptionWakeup, DEC); Serial.print(F("/"));  Serial.println(_OptionWakeup, BIN);
	Serial.print(F("OptionFEC (HEX/DEC/BIN)         : ")); Serial.print(_OptionFEC, HEX); Serial.print(F("/"));  Serial.print(_OptionFEC, DEC); Serial.print(F("/"));  Serial.println(_OptionFEC, BIN);
	Serial.print(F("OptionPower (HEX/DEC/BIN)       : ")); Serial.print(_OptionPower, HEX); Serial.print(F("/"));  Serial.print(_OptionPower, DEC); Serial.print(F("/"));  Serial.print(_OptionPower, BIN);Serial.print(" ");Serial.print(transmitPower[_OptionPower]);Serial.println("dbm");

	Serial.println("----------------------------------------");

}

```

knopserl, 2020-02-07:
@KrisKasprzak I also contacted EBYTE and here is the answer:

Dear Heinz, 

Thanks for choosing Ebyte. Sorry for the inconvenience. 

The E32 and E28 series don't have RSSI function because of hardware limitation. 
The E22 series support the RSSI function, please view this page to see various models of E22: http://www.ebyte.com/en/product-class.aspx?key=E22 

Please feel free to ask other questions about our products and i can also offer you discounted prices for your large orders. 
---------------------------------Best Regards,Sales Engineer:Robin HuEmail: sales06@ebyte.comPhone: +86-15817307997(whatsapp)Website: http://www.ebyte.com/en/Aliexpress：https://www.aliexpress.com/store/4836041Address:Innovation Center D347,4# XI-XIN road,High-tech district(west),Chengdu,Sichuan,China

KrisKasprzak, 2020-02-07:
i never heard back, bummer....
it would have been great to have power levels

Thanks,
Kris Kasprzak

closed 2020-10-04 by KrisKasprzak