KrisKasprzak/EBYTE #15: Set Sync word

onipot, 2020-04-08:
How can we set the sync word? There is no available method.

KrisKasprzak, 2020-04-08:
sync word? I don't recall seeing this in the data sheet. If you can point me to the documentation, i'll look at adding the methods

onipot, 2020-04-09:
You can't find anything related to it in the docs, but it's a parameter that SX1276 allows to change, and I didn't find a correspective in the EBYTE docs. Thought you knew something about this.
I am trying to make communicate a sx1276 (heltec lora v2) with an EBYTE 868T30D, and air data rate, parity bit and sync word must be the same. The problem is that: I can't find air data rate nor parity bit specs on the heltec and nothing about sync word on the EBYTE. 
Do you have any tips? I'm kinda out of ideas right now.

closed 2020-04-30 by onipot

clabnet, 2021-03-20:
+1, same issue.

renanqts, 2021-04-04:
@masc-it did you find a solution for this?

ivanboesing, 2021-05-02:
+2, same issue

luccareinehr, 2021-07-16:
+3

masc-it, 2021-07-17:
I tried them all, no solutions. Eventually I bought another EBYTE card and used LoRa_E32_Series_Library by xreef. If you want I can share some setup code :)

karfam, 2022-01-07:
> EBYTE card a

Hello Sir
if possible share some code.

KrisKasprzak, 2022-01-08:
what do you need that is different from the examples?
Thanks,
Kris Kasprzak

karfam, 2022-01-08:
Good Morning Kris
Sorry for my misunderstanding. 
I thought @masc-it  had some code so that EBYTE module can transmit a message a different Lora module manufacturer like TTGO-TBEAM.
But this is not case here :)

I have found some settings between EBYTE air data rate and Lora Settings but still no luck

0.3kbps - SF: 12 BW: 7 CR: 1
1.2kbps - SF: 11 BW: 8 CR: 1
2.4kbps - SF: 11 BW: 9 CR: 1
4.8kbps - SF: 8 BW: 8 CR: 2