KrisKasprzak/EBYTE #16: Minor issue and a question

tqueale, 2020-04-11:
In EBYTE.cpp at line 263 in Reset() comments you have "Per the data sheet just send 3x 0x4C to the unit" - this should be '0xC4' in place of the '0x4C'. The code is correct.

Does this Reset() restore factory defaults (those specified as defaults in the data sheet) or does it restore what you have saved with SaveParameters(PERMANENT)? The data sheet does not make this clear.

KrisKasprzak, 2020-04-12:
I have no idea what the datasheet indicates reset will do. sending 3 0XC4 will not reset the unit to the factory defaults. I'll contact the MFG and ask.

KrisKasprzak, 2020-12-06:
I have been playing around more with the reset function which is now a public function by the way it appears to clear the internal buffer of the transceiver. I am now using this function by own code to clear the buffer after programming in especially useful when trying to program with incoming data

closed 2020-12-06 by KrisKasprzak