KrisKasprzak/EBYTE #17: Print Parameters Error

cglkr97, 2020-04-17:
Hey Kris!

I have a breadboard Arduino that works with 8MHz external crystal on 3.3V. I use my E32-TTL-100 on this board. I set the Lora module's parameters via the parameter setting software created by EBYTE. I set the parameters successfully but when I upload your Transciever.PrintParameters() code, I get this,

> 14:13:20.610 -> Mode (HEX/DEC/BIN): 0/0/0
14:13:20.650 -> AddH (HEX/DEC/BIN): 0/0/0
14:13:20.690 -> AddL (HEX/DEC/BIN): 0/0/0
14:13:20.730 -> Sped (HEX/DEC/BIN): 0/0/0
14:13:20.730 -> Chan (HEX/DEC/BIN): 0/0/0
14:13:20.770 -> Optn (HEX/DEC/BIN): 0/0/0
14:13:20.810 -> Addr (HEX/DEC/BIN): 0/0/0
14:13:20.810 ->  
14:13:20.810 -> SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
14:13:20.879 -> SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
14:13:20.914 -> SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
14:13:20.948 -> OptionTrans (HEX/DEC/BIN)       : 0/0/0
14:13:20.983 -> OptionPullup (HEX/DEC/BIN)      : 0/0/0
14:13:21.052 -> OptionWakeup (HEX/DEC/BIN)      : 0/0/0
14:13:21.087 -> OptionFEC (HEX/DEC/BIN)         : 0/0/0
14:13:21.122 -> OptionPower (HEX/DEC/BIN)       : 0/0/0
14:13:21.155 -> ----------------------------------------

But I set the parameters to the following values which definitely not all 0s as you can see below.

[image]

In the previous issues, I see that you think this error might be caused by the MCU. My MCU seems to work fine because I've worked with it before and I check the wirings a hundred times and they seem fine also. Just to give you more insight into the problem, I've tried another library (https://github.com/xreef/LoRa_E32_Series_Library) and it also gave the same error of printing my parameters different from the ones I set using the EBYTE software. 

So why do you think this happens? Could you please give some more info on why this problem might be caused by the MCU? I'm doing my senior project on these modules and I'm stuck here. 
Thank you in advance!
C.

BJPatriotRacing, 2020-04-17:
Arduinos can be finicky, i've seen this a few times but can't isolate why. I tend to find Teensy (3v3 units work more reliably.  Are you sure you have Arduino TX connected to the EBYTE RX and the Arduino RX connected to the Ebyte TX? And MO, MO, Aux pins defined correctly? If i had a dollar for every time i inadvertently swapped AUX and TX :)

 I have a hunch it's a timing issue in setting pins or serial timeout. Try this
1) add the time out parameter and set to to say 5000 ms (default is 1000 ms)
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX), 5000;
2. in the .h file, increase the pinmode recover time to say 1000 ms
#define PIN_RECOVER 1000
3. try 4K7 pullups on AUX and TX
4. try power the module with Arduinos 5V0 ('ve tried mine on 5v0 and 3v3 and both seem to work

I just tested 4 different modules connected to a NANO, each powered with 5v0 and no pullups. Note, some have had a few parameters changed, so things like channel will not the factory setting.

Here's my serial dump of my PrintParameters() of an E50
----------------------------------------
Model no.: 50
Version  : 48
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): DC/220/11011100
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------

Here's my serial dump of my PrintParameters() of an E50
----------------------------------------
Model no.: 51
Version  : 47
Features : 1B
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): 0/0/0
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------

Here's my serial dump of my PrintParameters() of an E44
----------------------------------------
Model no.: 44
Version  : C
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1D/29/11101
Chan (HEX/DEC/BIN): 5/5/101
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 5/5/101
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------

Here's my serial dump of my PrintParameters() of an E44
----------------------------------------
Model no.: 53
Version  : 48
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 1/1/1
AddL (HEX/DEC/BIN): 1/1/1
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): F/15/1111
Optn (HEX/DEC/BIN): 40/64/1000000
Addr (HEX/DEC/BIN): 101/257/100000001
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 0/0/0
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------

I hope this helps

KrisKasprzak, 2020-04-17:
Arduinos can be finicky, i've seen this a few times but can't isolate why. I tend to find Teensy (3v3 units work more reliably. Are you sure you have Arduino TX connected to the EBYTE RX and the Arduino RX connected to the Ebyte TX? And MO, MO, Aux pins defined correctly? If i had a dollar for every time i inadvertently swapped AUX and TX :)

I have a hunch it's a timing issue in setting pins or serial timeout. Try this

add the time out parameter and set to to say 5000 ms (default is 1000 ms)
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX), 5000;
in the .h file, increase the pinmode recover time to say 1000 ms
#define PIN_RECOVER 1000
try 4K7 pullups on AUX and TX
try power the module with Arduinos 5V0 ('ve tried mine on 5v0 and 3v3 and both seem to work
I just tested 4 different modules connected to a NANO, each powered with 5v0 and no pullups. Note, some have had a few parameters changed, so things like channel will not the factory setting.

Here's my serial dump of my PrintParameters() of an E50
Model no.: 50
Version : 48
Features : 14

Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): DC/220/11011100
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN) : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN) : 0/0/0
OptionTrans (HEX/DEC/BIN) : 0/0/0
OptionPullup (HEX/DEC/BIN) : 1/1/1
OptionWakeup (HEX/DEC/BIN) : 0/0/0
OptionFEC (HEX/DEC/BIN) : 1/1/1
OptionPower (HEX/DEC/BIN) : 0/0/0
Here's my serial dump of my PrintParameters() of an E50
Model no.: 51
Version : 47
Features : 1B

Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): 0/0/0
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN) : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN) : 0/0/0
OptionTrans (HEX/DEC/BIN) : 0/0/0
OptionPullup (HEX/DEC/BIN) : 1/1/1
OptionWakeup (HEX/DEC/BIN) : 0/0/0
OptionFEC (HEX/DEC/BIN) : 1/1/1
OptionPower (HEX/DEC/BIN) : 0/0/0
Here's my serial dump of my PrintParameters() of an E44
Model no.: 44
Version : C
Features : 14

Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1D/29/11101
Chan (HEX/DEC/BIN): 5/5/101
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN) : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN) : 5/5/101
OptionTrans (HEX/DEC/BIN) : 0/0/0
OptionPullup (HEX/DEC/BIN) : 1/1/1
OptionWakeup (HEX/DEC/BIN) : 0/0/0
OptionFEC (HEX/DEC/BIN) : 1/1/1
OptionPower (HEX/DEC/BIN) : 0/0/0
Here's my serial dump of my PrintParameters() of an E44
Model no.: 53
Version : 48
Features : 14

Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 1/1/1
AddL (HEX/DEC/BIN): 1/1/1
Sped (HEX/DEC/BIN): 18/24/11000
Chan (HEX/DEC/BIN): F/15/1111
Optn (HEX/DEC/BIN): 40/64/1000000
Addr (HEX/DEC/BIN): 101/257/100000001

SpeedParityBit (HEX/DEC/BIN) : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN) : 0/0/0
OptionTrans (HEX/DEC/BIN) : 0/0/0
OptionPullup (HEX/DEC/BIN) : 1/1/1
OptionWakeup (HEX/DEC/BIN) : 0/0/0
OptionFEC (HEX/DEC/BIN) : 0/0/0
OptionPower (HEX/DEC/BIN) : 0/0/0
I hope this helps

cglkr97, 2020-04-18:
Hey Kris!

Thank you for taking the time to write me a detailed reply. This is gonna sound so stupid but I did swap the Tx and Rx which is not entirely my fault. I have been trying to use the other library that I gave you the link to. And in the beginning of his examples he says that we should do so.

Anyways, thanks to you I remembered to question that decision. Now when I upload Transciever.PrintParameters(), I see the parameters correctly and most importantly I made my first transmission! So yay!
Thanks again!
Best,
C.

KrisKasprzak, 2020-04-18:
Glad it worked. When you first filed this issue, i re-tested my lib with a NANO and got all zeros as well. I thought i was able to finally reproduce the issue so  I tied pull-ups, series resistors, etc and nothing worked, Then I checked my wiring--guess what... Tx and Aux were flipped. Silly me. Once wired correctly, the modules fired right up and gave the correct parameters during print.
Feel no shame, this stuff is unforgiving and not as easy as the YouTube videos claim :)
Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

closed 2020-12-06 by KrisKasprzak