KrisKasprzak/EBYTE #18: E32868T20D configuration problem

onipot, 2020-04-30:
I have a sender (Arduino Mega) and a Receiver (esp32, heltec v2).

These are the params:

```
MEGA

----------------------------------------
Model no.: 45
Version  : D
Features : 14

Mode (HEX/DEC/BIN): 0/0/0
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1A/26/11010
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 47/71/1000111
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 2/2/10
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 3/3/11
----------------------------------------
```

```
ESP32
----------------------------------------
Model no.: 45
Version  : D
Features : 14

Mode (HEX/DEC/BIN): 0/0/0
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1A/26/11010
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 47/71/1000111
Addr (HEX/DEC/BIN): 0/0/0
      
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 2/2/10
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 3/3/11
----------------------------------------
```
Everything seems the same on both the modules, but the receiver doesn't receive anything at all.

Receiver loop code
```
if (ESerial.available()) {

    Transceiver.GetStruct(&MyData, sizeof(MyData));

    Serial.print("Count: "); Serial.println(MyData.c);

  }
```
Sender loop code
```
  strcpy(MyData.c , "test");

  Transceiver.SendStruct(&MyData, sizeof(MyData));

  Serial.print("Sending: "); Serial.println(MyData.c);
  delay(500);
```

The structure sent:
```
struct DATA {
  
  char c[10];

};
```

KrisKasprzak, 2020-05-04:
I've had issues in sending structs between 2 different MCU's as well. I believe it's due to how each compiler packs the data--odd it can't match a simple char.  Some claim you can only use uint16_t between different boards.
Try using EasyTransfer.h (that lib has a .h for hardware serial and software serial application). I'm having to use that lib on a current project and it works. I send / receive data using EasyTransfer and the EBYTE lib for module management.
madsci1016/Arduino-EasyTransfer

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
madsci1016/Arduino-EasyTransfer

An Easy way to Transfer data between Arduinos. Contribute to madsci1016/Arduino-EasyTransfer development by crea...
 |

 |

 |

Thanks,
Kris

onipot, 2020-06-23:
@KrisKasprzak can you provide some samples using EasyTransfer? I think it would be useful for lots of people :)

vindolin, 2020-09-16:
@onipot I have the same problem with two ESP32 MCUs.

Transceiver.PrintParameters() prints the default values on both modules.

```
----------------------------------------
Model no.: 45
Version  : D
Features : 14
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1A/26/11010
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 2/2/10
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
```
Have you tried just sending/receiving single bytes instead of the struct, just to see, if the transmission works?

I first tried `Transceiver.SendStruct()`, then the EasyTransfer library and finally just `Transceiver.SendByte(58)` in the TX loop, and on the RX side:
```
if (Serial2.available()) {
    Serial.println(Transceiver.GetByte());
}
```
But all to no avail :(

I'm ordering some new modules now to see if one is broken.

KrisKasprzak, 2020-09-16:
I have tested sending recieving struct and bytes.
I'll have another look

Thanks,
Kris

closed 2020-12-06 by KrisKasprzak