KrisKasprzak/EBYTE #20: E22 commands difference

aleksander1084, 2020-06-10:
Hi!
I have a question, have you been testing your library with E22 modules?
I think there might be an issue because they have different command formats and responds.
As well as for E22 programming mode is different, for E32 (M0 M1) was (1 1), for E22 it's (0 1).

Sorry that I can't test it myself since since my modules are still being shipped what takes years this days.

Thanks in advance for any help,
Aleksander

Ashraff93, 2021-02-17:
Any solution when using E22 instead of E32 since i have same problem about this?

aleksander1084, 2021-02-17:
Based on Krise's library I had build my own dedicated for E22, there is several differences between E22 and E32 + there is several mistakes in a Datasheet that I have reported to E-byte but haven't seen them fixed.

Dukat-Gul, 2021-05-03:
> Based on Krise's library I had build my own dedicated for E22, there is several differences between E22 and E32 + there is several mistakes in a Datasheet that I have reported to E-byte but haven't seen them fixed.

Just bumping this to see if @aleksander1084 has had any responses back from e-byte? And/or are you willing to share (please) the work you have done to make your dedicated E22 build? I am keen to give it a try on my E22s. Cheers

Raniita, 2021-05-05:
> Based on Krise's library I had build my own dedicated for E22, there is several differences between E22 and E32 + there is several mistakes in a Datasheet that I have reported to E-byte but haven't seen them fixed.

I add to @Dukat-Gul comment as to whether it would be possible for you (@aleksander1084 ) to share with us the modifications you have made to make the E22 work. I am currently designing a project, with e32, but im looking at modules with RSSI feature (e22 or e220). It would be very interesting to see how you have implemented it. 

Thank you all!! :D

closed 2021-08-30 by KrisKasprzak