KrisKasprzak/EBYTE #21: Channel spacing - 1 MHz ! 433MHz

Cloolalang, 2020-07-13:
Hi guys, am I incorrect with my discovery that my E23-TTL-100 433MHz lora Uart radio only does 1MHz channels?!
So ISM band is 433.050 to 434.790 MHz where I am,
The Bandwidth for Lora Phy at 0.3Kb on-air is 125KHz.
That would allow for:
433.000 Channel 1
433.125 Channel 2 - not available on this device.
433.250 Channel 3  - not available on this device.
433.375 Channel 4 - not available on this device.
455.500 Channel 5 -not available on this device.

.....

Really? only 2 ISM 433 MHz usable frequencies?

Please tell me that these is better firmware for operating a finer frequency selection.

Please! Please!
P

KrisKasprzak, 2020-07-13:
I would recommend you read the data sheet
E32-433T20DC-成都亿佰特电子科技有限公司

| 
| 
|  | 
E32-433T20DC-成都亿佰特电子科技有限公司

赛门仕博,www.symansbon.cn

 |

 |

 |

that unit is 410 mhz to 441 mhz. the channel sets the frequency within that range in 1mhz increments hence 31 different channels

Thanks,
Kris Kasprzakkris.kasprzak@yahoo.com

Cloolalang, 2020-07-14:
Hi Kris, Thank-you for taking a look!
OK so yes, 
I am correct, I can only legally use:
channel 24 (434.000 MHz) assuming I reduce the output power to low enough and keep the duty cycle correct.
Also I see that there does not appear to be any CAD mode implemented for the module (this is reduce collisions like "listen-before-transmit").
The other 30 RF channels cannot be used without a license.
P

KrisKasprzak, 2020-07-14:
where is it written that the only legal frequency is 434 mhz?

Cloolalang, 2020-07-14:
Hi Kris.
https://www.ofcom.org.uk/__data/assets/pdf_file/0028/84970/ir-2030.pdf
See pages 21 & 22.
p

KrisKasprzak, 2020-12-06:
The channel frequencies are based on the chip not much you can do about it

closed 2020-12-06 by KrisKasprzak