KrisKasprzak/EBYTE #24: Erro with Arduino NANO Every

Furanbelius, 2020-08-07:
When doing tests with the Arduino NANO Every card I have a constant error with the EBYTE library, I tried the Arduino and Teensy examples, since the latter has more than one serial port like the card with which I am testing, but not I have gotten it to work.
Reading about the Arduino NANO Every, the SoftwareSerial library does not work nor has it been ported since the serial port available to work is Serial1 and it is separate from the one used for USB communication.
The error is the following, when I try the send sketch:

Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), Tarjeta:"Arduino Nano Every, None (ATMEGA4809)"

In file included from C:\xXx\AppData\Local\Temp\arduino_modified_sketch_750088\Send.ino:21:0:

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:175:15: error: expected ')' before '*' token

  EBYTE(Stream *s, uint8_t PIN_M0 = 4, uint8_t PIN_M1 = 5, uint8_t PIN_AUX = 6, unsigned long ReadTimeout = 1000);

               ^

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:270:2: error: reference to 'Stream' is ambiguous

  Stream*  _s;

  ^~~~~~

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:169:7: note: candidates are: class Stream

 class Stream;

       ^~~~~~

In file included from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Client.h:22:0,

                 from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/ArduinoAPI.h:29,

                 from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/Arduino.h:23,

                 from sketch\Send.ino.cpp:1:

D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Stream.h:50:7: note:                 class arduino::Stream

 class Stream : public Print

       ^~~~~~

In file included from C:\Users\Francis\AppData\Local\Temp\arduino_modified_sketch_750088\Send.ino:21:0:

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:271:2: error: reference to 'Stream' is ambiguous

  Stream*  _TD;

  ^~~~~~

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:169:7: note: candidates are: class Stream

 class Stream;

       ^~~~~~

In file included from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Client.h:22:0,

                 from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/ArduinoAPI.h:29,

                 from D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/Arduino.h:23,

                 from sketch\Send.ino.cpp:1:

D:\XxX\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Stream.h:50:7: note:                 class arduino::Stream

 class Stream : public Print

       ^~~~~~

Send:46:51: error: no matching function for call to 'EBYTE::EBYTE(UartClass*, int, int, int)'

 EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX);

                                                   ^

In file included from C:\Users\Francis\AppData\Local\Temp\arduino_modified_sketch_750088\Send.ino:21:0:

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note: candidate: EBYTE::EBYTE()

 class EBYTE {

       ^~~~~

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note:   candidate expects 0 arguments, 4 provided

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note: candidate: constexpr EBYTE::EBYTE(const EBYTE&)

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note:   candidate expects 1 argument, 4 provided

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note: candidate: constexpr EBYTE::EBYTE(EBYTE&&)

D:\XxX\Arduino\libraries\EBYTE-master/EBYTE.h:171:7: note:   candidate expects 1 argument, 4 provided

exit status 1

no matching function for call to 'EBYTE::EBYTE(UartClass*, int, int, int)'

KrisKasprzak, 2020-08-08:
I think the .h and .cpp got out of synch, try downloading again. I just tried on a NANO and it compiled just fine.

Furanbelius, 2020-08-11:
The test was with just an Arduino Nano? because im using an Arduino Nano Every, that use the ATMEGA4809, and with that i cant solve the issue to work

KrisKasprzak, 2020-08-11:
I have never used that MCU, what pins are you using?

Furanbelius, 2020-08-11:
Tx and RX, those are the TX1(PC4) and RX1(PC5) in the  ATMEGA4809, for the usb communication TX0 and RX0 are been used.

more info about the board: https://store.arduino.cc/usa/nano-every

closed 2021-03-21 by KrisKasprzak

Guy1000, 2022-12-31:
Did anyone every get it figured out? I am having the exact same problem.