KrisKasprzak/EBYTE #27: Transceiver.PrintParameters(); returns only 0 values on a ESP32 / 868T20D

vindolin, 2020-09-15:
I get the following output when calling Transceiver.PrintParameters(); 
```
----------------------------------------
Model no.: 0
Version  : 0
Features : 0
 
Mode (HEX/DEC/BIN): 0/0/0
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 0/0/0
Chan (HEX/DEC/BIN): 0/0/0
Optn (HEX/DEC/BIN): 0/0/0
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 0/0/0
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 0/0/0
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
```
Here's the code I'm using:

```C++
#include "EBYTE.h"

#define PIN_RXD2 16
#define PIN_TXD2 17

#define PIN_M0 19
#define PIN_M1 22
#define PIN_AX 21

// 868T20D

EBYTE Transceiver(&Serial2, PIN_M0, PIN_M1, PIN_AX);  // using hardware serial2 on RXD2/TXD2

void setup() {
    Serial.begin(9600);
    Serial2.begin(9600, SERIAL_8N1, PIN_RXD2, PIN_TXD2);

    Transceiver.init();
    Transceiver.PrintParameters();
}

void loop() {
}

```

KrisKasprzak, 2020-09-15:
What MCU are you using and what are your pin connections?

if you are using a teensy 3.2, Serial2 is Rx2=7, and Tx2=8
also try simple Serial2.begin(9600);

vindolin, 2020-09-15:
I'm using an ESP32.

With the ESP32 Core libraries one can choose the pins where the hardware serials should run on.

KrisKasprzak, 2020-09-15:
I have an ESP32. i'll give it a try

KrisKasprzak, 2020-09-16:
Thomas,
Gooten moorgen!
I connected my Espressif ESP32 to an E53-TTL-100--which I believe is the same as yours but an old model number.
Running your code (with a few Serial.println's)
/*
pinouts
ESP32       EBYTE 19           MO 22           M1 21           RXD 16           TXD 17           AUX 3v3          VCC GND          GND
 */
#include "EBYTE.h"
#define PIN_RXD2 16#define PIN_TXD2 17#define PIN_M0 19#define PIN_M1 22#define PIN_AX 21
EBYTE Transceiver(&Serial2, PIN_M0, PIN_M1, PIN_AX);  // using hardware serial2 on RXD2/TXD2
void setup() {  Serial.begin(9600);  Serial2.begin(9600, SERIAL_8N1, PIN_RXD2, PIN_TXD2);  Serial.println("starting");
// try adding this...the lib may init before your module has stable power--yes i'm reaching here// delay(1000);
  Transceiver.init();  Transceiver.PrintParameters();  Serial.println("looping 2");}
void loop() {

}

produces this on the serial monitor

starting----------------------------------------Model no.: 53Version  : 48Features : 14 Mode (HEX/DEC/BIN): C0/192/11000000AddH (HEX/DEC/BIN): 0/0/0AddL (HEX/DEC/BIN): 0/0/0Sped (HEX/DEC/BIN): 18/24/11000Chan (HEX/DEC/BIN): C/12/1100Optn (HEX/DEC/BIN): 44/68/1000100Addr (HEX/DEC/BIN): 0/0/0 SpeedParityBit (HEX/DEC/BIN)    : 0/0/0SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0OptionTrans (HEX/DEC/BIN)       : 0/0/0OptionPullup (HEX/DEC/BIN)      : 1/1/1OptionWakeup (HEX/DEC/BIN)      : 0/0/0OptionFEC (HEX/DEC/BIN)         : 1/1/1OptionPower (HEX/DEC/BIN)       : 0/0/0----------------------------------------looping 2

I verified the EBYTE library results with the hardware programmer that came with my transceiver and the data read matches what is on the module.
1. are you sure you have the RX and TX lines reversed and connected to the correct pins?2. are you sure all other pins are correct?
this may sound silly to be asking but my first attempt failed as I had MO on the wrong pin.

Thanks,
Kris

vindolin, 2020-09-16:
Good morning and thanks for your quick answer.
Yes I double checked the pins and also tried different pins on the other side of the module (34-38).
I tried even longer delays but with the same result.
Connected to a FTDI adapter and using the Ebyte RF Setting tool, I can read the configuration just fine.
I will try to query the device ID directly without the library later and see if I get anything back.

Thanks
Thomas

vindolin, 2020-09-16:
After pulling my hairs for some hours I took my multimeter and checked all the jumper cables.
Turned out, one was broken🤦‍♂️
German proverb:  "Der Teufel ist ein Eichhörnchen"

All seems to work now :)
```
----------------------------------------
Model no.: 45
Version  : D
Features : 14

Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1A/26/11010
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 2/2/10
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
```

Sorry for stealing your time :(

closed 2020-09-16 by vindolin

KrisKasprzak, 2020-09-16:
Glad all is working for you.Thanks,Kris Kasprzak256-348-0953Kris.kasprzak@yahoo.com

  0 values on a ESP32 / 868T20D (#27)