KrisKasprzak/EBYTE #33: E32-433t30d .. rssi or snr ???

farmaceuta777, 2021-02-02:
Hello Kris ..
is there any possibility to get RSSI or SNR values ​​??
I would like to be able to estimate the possible capacity because I plan to use the modules in a mobile vehicle ... and this value would be very necessary ...
please answer :-)
best regards..

KrisKasprzak, 2021-02-02:
According to the data sheet this data is not exposed in any form or fashion. I don't know if any way to get it out of the chip.Thanks,Kris Kasprzak

DedePele, 2021-02-10:
Hello Kris, Thanks you for your post and support. Browsing on internet I found this post. I also just sent a message to Me Robin Hu (eByte).

I am beginer at LoRa and I got some UART E32 433 1W.

We want to make a very simple link point to point and now I can not understand how to monitor the link performance without RSSI ou SNR...

I do not understand if it's true because this is the main information... Maybe we can get that by some library? 

I will be glad if you could help us at an alternative way to monitor the signal quality.

Thanks!

BR

André

KrisKasprzak, 2021-02-10:
these parameters are not exposed, there is nothing I can do
Thanks,

closed 2021-08-30 by KrisKasprzak