KrisKasprzak/EBYTE #34: Testing with NodeMCU 8266

clabnet, 2021-03-20:
Hi, 
I have two NodeMCU and a pair of E32868TD.
Using yor library I can't connect correctly the E32 with MCU.
I have tried much many options, first using assign -1 to M0, M1, and AUX. No working.
The last I used complete GPIO assigning, and also no working.
```
#define PIN_RX  10
#define PIN_TX  9
#define PIN_M0  4
#define PIN_M1  5
#define PIN_AUX 14
```

This is my entire receiver code, it is the same as is your sample, plus led :

```

void setup()
{

  Serial.begin(115200);

  pinMode(LED_BUILTIN, OUTPUT);    // Initialize the LED_BUILTIN pin as an output
  digitalWrite(LED_BUILTIN, HIGH); // Turn the LED off by making the voltage HIGH    }

  ESerial.begin(9600);
  Serial.println("Starting Receiver");

  // this init will set the pinModes for you
  Transceiver.init();

  // all these calls are optional but shown to give examples of what you can do

  // Serial.println(Transceiver.GetAirDataRate());
  // Serial.println(Transceiver.GetChannel());

  // Transceiver.SetAddressH(1);
  // Transceiver.SetAddressL(0);
  // Chan = 5;
  // Transceiver.SetChannel(Chan);
  // save the parameters to the unit,
  // Transceiver.SaveParameters(PERMANENT);

  // you can print all parameters and is good for debugging
  // if your units will not communicate, print the parameters
  // for both sender and receiver and make sure air rates, channel
  // and address is the same
  Transceiver.PrintParameters();
}

void loop()
{

  // if the transceiver serial is available, proces incoming data
  // you can also use Transceiver.available()

  if (ESerial.available())
  {

    digitalWrite(LED_BUILTIN, LOW); // Turn the LED on by making the voltage LOW
    delay(50);
    digitalWrite(LED_BUILTIN, HIGH); // Turn the LED off by making the voltage HIGH    }
    delay(50);

    // i highly suggest you send data using structures and not
    // a parsed data--i've always had a hard time getting reliable data using
    // a parsing method

    Transceiver.GetStruct(&MyData, sizeof(MyData));

    // dump out what was just received
    Serial.print("Count: ");
    Serial.println(MyData.Count);
    Serial.print("Bits: ");
    Serial.println(MyData.Bits);
    Serial.print("Volts: ");
    Serial.println(MyData.Volts);
    // if you got data, update the checker
    Last = millis();
  }
  else
  {
    delay(50);
  }
}
```

The result is always 
```
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Starting Receiver
trying: 0
trying: 1
trying: 2
trying: 3
trying: 4
----------------------------------------
Model no.: 0
Version  : 0
Features : 0

Mode (HEX/DEC/BIN): 0/0/0
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 0/0/0
Chan (HEX/DEC/BIN): 0/0/0
Optn (HEX/DEC/BIN): 0/0/0
Addr (HEX/DEC/BIN): 0/0/0

SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 0/0/0
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 0/0/0
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
```

It seems not have the connection between NODEMCU and E32. The RX is powered by 3.3, common as the same as ground of MCU. UI have changed a lot of wires and I have tested all what I could. 

Any chance to run ?

Thanks, regards

Claudio

KrisKasprzak, 2021-03-21:
Not sure of your pin definition is the GPIO pin of the pin number. Also I'm not sure you can use any GPIO pin for serial. This worked for me

/*
  This example shows how to connect to an EBYTE transceiver
  using an ESP8266
  This code for for the sender
  connections
  Module      ESP6266
  M0          GPIO14 (pin D5)
  M1          GPIO5  (pin D6)
  Rx          GPIO2  (pin D1)
  Tx          GPIO3  (Pin D2)
  Aux         GPIO6  (pin D0)
  Vcc         3V3
  Gnd         Gnd
*/

#include <SoftwareSerial.h>
#include "EBYTE.h"

/*
WARNING: IF USING AN ESP8266
DO NOT USE THE PIN NUMBERS PRINTED ON THE BOARD
YOU MUST USE THE ACTUAL GPIO NUMBER
*/

#define PIN_RX 14   //D5 on the board
#define PIN_TX 12   //D6 on the board

#define PIN_M0 5    //D1 on the board
#define PIN_M1 4    //D2 on the board
#define PIN_AX 16   //D0 on the board

// i recommend putting this code in a .h file and including it
// from both the receiver and sender modules

// these are just dummy variables, replace with your own
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};

int Chan;
DATA MyData;
int i;

// you will need to define the pins to create the serial port
SoftwareSerial ESerial(PIN_RX, PIN_TX);

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX);

closed 2021-03-21 by KrisKasprzak

KrisKasprzak, 2021-03-21:
My initial response had a typo but has been updated. Make sure MCU Rx connects to EBTYE Tx and MCU Tx connects to EBTYE Rx.

#define PIN_RX 14   //D5 on the board (connect this to the EBYTE TX pin)
#define PIN_TX 12   //D6 on the board (connect this to the EBYTE RX pin)

#define PIN_M0 5    //D1 on the board (connect this to the EBYTE M0 pin)
#define PIN_M1 4    //D2 on the board (connect this to the EBYTE M1 pin)
#define PIN_AX 16   //D0 on the board (connect this to the EBYTE AUX pin)