KrisKasprzak/EBYTE #35: Hardware serial

PartyLime99, 2021-04-30:
Hi,

Sorry if this is the wrong place to post this. I'm struggling to find a good place to ask my question. 

Your library has been great while trying to set up a laser gate timing system. I have 2 Arduino Nanos sending and an Uno receiving. I am now onto adding a screen to my setup and have switched to an Adafruit Metro M0 rather than the Uno. The M0 does not allow a software serial.

You mention in your description "Some MCU such as the Teensy, and ESP32 do NOT allow the use of SoftwareSerial to create a communications port. No worries, just hard wire the EBTYE to a dedicated UART port (pin 0 and pin 1 on a teensy 3.2 for Serial1." however, I am very new to programming and can't figure out how to set this up. What do I replace  `SoftwareSerial loraSerial (2,3); //TX, RX` with? I have tried replacing SoftwareSerial with Serial1 but it says "Serial1 does not name a type".

I have my basic code below which works as a receiver on the uno

`#include <SoftwareSerial.h>
#include "EBYTE.h"

//If using direct wiring, use these pin definitions
//#define PIN_M0 4 
//#define PIN_M1 5 
//#define PIN_AX 6

SoftwareSerial loraSerial (2,3); //TX, RX
EBYTE Transceiver(&loraSerial); //PIN_M0, PIN_M1, PIN_AX);

void setup() {
  
  Serial.begin(9600);
  loraSerial.begin(9600);
  Transceiver.init();
  Transceiver.SetMode(MODE_NORMAL);
  Transceiver.SetAirDataRate(ADR_9600);
  Transceiver.SetTransmitPower(OPT_TP17);
  
}

void loop() {
  
   if (loraSerial.available()>0){
    String input = loraSerial.readString();
    Serial.print("I received: "); Serial.println(input);
   }

}
`

KrisKasprzak, 2021-04-30:
From what I see that unit has only 1 hardware serial port Pin0=Rx and Pin1=Tx, This means you will have to connect the EBYTE Rx to pin1 and EBYTE Tx to pin0--however you will not be able to program the unit when these pins are connected to the EBYTE. 
// SoftwareSerial loraSerial (2,3); //TX, RX
EBYTE Transceiver(Serial); //PIN_M0, PIN_M1, PIN_AX);

Basically  unconnect the EBYTE, program the unit and connect the EBYTE back up--you can't use Serial monitor when EBYTE is connected, so debugging is a pain.

I have a similar situation and i put jumpers on my PCB board for "program" and "run"

KrisKasprzak, 2021-04-30:
> Thank you so much for the fast reply!
>
> So I have done as you say and I get the following error while compiling:
> Arduino: 1.8.13 (Windows 10), Board: "Adafruit Metro M0 Express, Small (-Os) (standard), Arduino, Off"
>
> Metro_LoRa_Receive_2:12:25: error: no matching function for call to 'EBYTE::EBYTE(Serial_&)'
>
>    12 | EBYTE Transceiver(Serial); //PIN_M0, PIN_M1, PIN_AX);
>
>       |                         ^
>
> In file included from C:\Users\benf_\Documents\Arduino\Metro_LoRa_Receive_2\Metro_LoRa_Receive_2.ino:2:
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:181:2: note: candidate: 'EBYTE::EBYTE(Stream*, uint8_t, uint8_t, uint8_t)'
>
>   181 |  EBYTE(Stream *s, uint8_t PIN_M0 = 4, uint8_t PIN_M1 = 5, uint8_t PIN_AUX = 6);
>
>       |  ^~~~~
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:181:16: note:   no known conversion for argument 1 from 'Serial_' to 'Stream*'
>
>   181 |  EBYTE(Stream *s, uint8_t PIN_M0 = 4, uint8_t PIN_M1 = 5, uint8_t PIN_AUX = 6);
>
>       |        ~~~~~~~~^
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note: candidate: 'constexpr EBYTE::EBYTE(const EBYTE&)'
>
>   177 | class EBYTE {
>
>       |       ^~~~~
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note:   no known conversion for argument 1 from 'Serial_' to 'const EBYTE&'
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note: candidate: 'constexpr EBYTE::EBYTE(EBYTE&&)'
>
> C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note:   no known conversion for argument 1 from 'Serial_' to 'EBYTE&&'
>
> exit status 1
>
> no matching function for call to 'EBYTE::EBYTE(Serial_&)'
>
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>
> I see how the no serial for debugging will be a nightmare. Hopefully the simplicity of the project will help. Maybe I can debug by writing on the screen attached to it.

you must pass in the other arguments for the following line...not sure why I commented it out in my first reply...
ChangeEBYTE Transceiver(Serial); //PIN_M0, PIN_M1, PIN_AX);
to
EBYTE Transceiver(Serial, PIN_M0, PIN_M1, PIN_AX);
Sorry....

PartyLime99, 2021-05-01:
I'm still getting the same error:

```
Metro_LoRa_Receive_2:12:50: error: no matching function for call to 'EBYTE::EBYTE(Serial_&, int, int, int)'
   12 | EBYTE Transceiver(ESerial, PIN_M0, PIN_M1, PIN_AX);
      |                                                  ^
In file included from C:\Users\benf_\Documents\Arduino\Metro_LoRa_Receive_2\Metro_LoRa_Receive_2.ino:2:
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:181:2: note: candidate: 'EBYTE::EBYTE(Stream*, uint8_t, uint8_t, uint8_t)'
  181 |  EBYTE(Stream *s, uint8_t PIN_M0 = 4, uint8_t PIN_M1 = 5, uint8_t PIN_AUX = 6);
      |  ^~~~~
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:181:16: note:   no known conversion for argument 1 from 'Serial_' to 'Stream*'
  181 |  EBYTE(Stream *s, uint8_t PIN_M0 = 4, uint8_t PIN_M1 = 5, uint8_t PIN_AUX = 6);
      |        ~~~~~~~~^
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note: candidate: 'constexpr EBYTE::EBYTE(const EBYTE&)'
  177 | class EBYTE {
      |       ^~~~~
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note:   candidate expects 1 argument, 4 provided
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note: candidate: 'constexpr EBYTE::EBYTE(EBYTE&&)'
C:\Users\benf_\Documents\Arduino\libraries\EBYTE-master/EBYTE.h:177:7: note:   candidate expects 1 argument, 4 provided
exit status 1
no matching function for call to 'EBYTE::EBYTE(Serial_&, int, int, int)'
```

I think instead I will use a nano with the ebyte module to recieve the message and set up I2C between the that and the Metro M0 to forward it on. I think this should simplify things for me and work for what I need.

KrisKasprzak, 2021-05-02:
you need to pass in Serial and NOT ESerial.
EBYTE Transceiver(Serial, PIN_M0, PIN_M1, PIN_AX);

KrisKasprzak, 2021-08-30:
Sorry, needs an ampersand.
EBYTE Transceiver(&Serial); //PIN_M0, PIN_M1, PIN_AX);

closed 2021-08-30 by KrisKasprzak