KrisKasprzak/EBYTE #39: E32-900T20D

Lucamenghini, 2021-08-19:
Hi thanks for your solution!
I tested the new E32-900T20D no changing parameters in your example.
The sender sends messages every 1000ms but the receiver receives 1 message yes and 1 not.
the sender is ... 32... 33... 34... 35... 36
the other receives 32... ... 34... ... 36... ... 38 and so on
why?
thenk you
bye

KrisKasprzak, 2021-08-19:
Seems odd but if air data rate is super slow, you may get missed receives.

Need more info
1. what MCU for sender and reciever
2. antennas attached?
3. distance between
4. what is air data rate?
5. my example sends a struct with count/bits/volts but you are only getting counts?
6. what does Transceiver.PrintParameters(); display in serial monitor?
7. anything connected to your MCU that may be sucking up time?

Note. you only need this lib if you need to change the default settings. Using 

ESerial.write((uint8_t*) &Data, PacketSize );

 ESerial.readBytes((uint8_t*)& MyData, (uint8_t) sizeof(MyData));

Lucamenghini, 2021-08-19:
Thank you Kris for your quick answer.
ESP32 Wroom 32 sender and receiver
the same antenna in sender and receiver: a 20dbl 868mhz
about 10 meters distance
the air data rate is the default (i have no setted it)
yes only get count

these are the parameters on esp32 setup going ( and then you can see te receiving data from the sender that was not restarted and continue to send). at the sender side i have another computer connected with the monitor and I see the trasmissions are regular (:
----------------------------------------
Model no.: 0
Version  : 45
Features : 32
 
Mode (HEX/DEC/BIN): C0/192/11000000
AddH (HEX/DEC/BIN): 0/0/0
AddL (HEX/DEC/BIN): 0/0/0
Sped (HEX/DEC/BIN): 1A/26/11010
Chan (HEX/DEC/BIN): 6/6/110
Optn (HEX/DEC/BIN): 44/68/1000100
Addr (HEX/DEC/BIN): 0/0/0
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 3/3/11
SpeedAirDataRate (HEX/DEC/BIN)  : 2/2/10
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 1/1/1
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 1/1/1
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
Searching: 
Searching: 
Count: 16608
Bits: 0
Volts: 0.00
Count: 16609
Bits: 0
Volts: 0.00
Count: 16612
Bits: 0
Volts: 0.00
Count: 16613
Bits: 0
Volts: 0.00
Count: 16616
Bits: 0
Volts: 0.00
Count: 16617
Bits: 0
Volts: 0.00
Count: 16620
Bits: 0
Volts: 0.00
Count: 16621

No other wires connected to sender and receiver cpu

thank you for your support
Best regards

Luca M

KrisKasprzak, 2021-08-20:
Per the data sheet you are transmitting at 2.4K. with that small amount of data I would expect the transmission to not lose any data, however, in my experience long distances take longer to transmit. My min rate is usually 9600 baud
Let's try to isolate the issue with an easy test...
First let's transmit less frequently. in the sender code, find line number 104 or so change delay(1000); to delay(3000); If that fixes it, then you can keep that fix or speed up the air data rate
To speed up the air data rate in setup add the following code, once run, you can comment it outTransceiver.SetAirDataRate( ADR_9600); // 9600 baud rate
Transceiver.SaveParameters(PERMANENT);
then verify by calling Transceiver.PrintParameters();

Verify that  SpeedAirDataRate (HEX/DEC/BIN) : 4/4/100and change delay(3000); back to delay(1000);

Thanks,

Kris Kasprzak
***@***.***

Lucamenghini, 2021-08-22:
Hi Kris thank you for your support! I changed the parameter:
Transceiver.SetAirDataRate( ADR_300);
in transmitter & receiver and the trasmission doesn't lose packet now.
The delay is 3000ms, the distance is about 10meters, a20dbi antenna on the transmitter and a small (4cm) antenna on receiver.
So solved
Thank you very much.... now i've got to battle with the last issue (i2c channel used by the E32900T20D and the SHT30 sensor, the last issues I recorded here.
bye

Luca M