KrisKasprzak/EBYTE #46: Specify the destination address or the channel in the sending

CodingByJerez, 2021-12-17:
Hello,
I don't know much about low-level programming language :/ it's all new to me.
Is it possible to specify the destination address or the channel only for sending a message ?

KrisKasprzak, 2021-12-17:
I’m not sure what you mean if you’re talking about sending an address for the device to then listen to the answer is no address in channel have to be set correctly for the device to hear any incoming messages.

CodingByJerez, 2021-12-17:
For example, let's say I have 5 devices each with a LORA card each identified by a different address and all in the same Chanel
 Is it possible to send the message only To another device is not at all thanks to the address specification at the time of sending the message?

KrisKasprzak, 2021-12-17:
In that case your transmitter will just have to send a different address to each of the receiving devices. All can be on the same channel. I have the same set up we’re all units are on the same channel, and the same address I just sent along a ID so the sender and or receiver know which device it was sent from.

CodingByJerez, 2021-12-19:
ok, I'm going to do it like that so I thought we could specify it on the menu with an address, I thought I had read that in their instructions.
Thank you a lot :)

closed 2021-12-19 by CodingByJerez