KrisKasprzak/EBYTE #48: error transmitting structure

Lucamenghini, 2022-01-23:
Hi! I wrote here in past about issues on parallel use of a E32900t20d and a SHT30 (they use I2C), I solved using I2C on others declared PINs.
My issue now is about the packet transmission structure.
I modified it into:

struct DATA {
  unsigned long Count;
  float Temperatura;
  float Umidita;
  String Uuid;
};

It is composed by 2 floats and a string.
The floats are ok, they are received with no problem but the string is sended for example with this  value
9C9C1FE20168

and it is received with strange characters.
What could be the problem?
The structure is the same in the sending sketch and in the receiving sketch.

Thank you for your great job and your support!
Bye

Luca Menghini

KrisKasprzak, 2022-01-23:
You should never use strings, for anything. Try using char’s

> On Jan 23, 2022, at 5:00 AM, Lucamenghini ***@***.***> wrote:
> 
> SHT30

KrisKasprzak, 2022-03-19:
string is an undefined length, char [29] cat; for example is known

closed 2022-03-19 by KrisKasprzak