KrisKasprzak/EBYTE #50: Two different MCUs

Yousefff1, 2022-03-19:
Hi!
I'm having issues transmitting data using two E220 modules but with different MCUs, in this case ESP8266 (receiver) and Atmega328P-AU (Transmitter). It seems floats transmit just fine, but Integers show up either as random (large) numbers or zero.

I'm using the barebone receive and transmit sketches examples from the library.

Any ideas?

KrisKasprzak, 2022-03-19:
Are you sending everything in a strict? If so it’s possible due to the fact that different MCU will compile struct differently and pack them differently. ESP32 32-bit device and the other is an eight. This probably will not work but try using a neutral data type try uin16_t. If that doesn’t work you’ll have to use a library called easy transfer. Let me know if you need an example on how to use that library

Thanks.
Kris Kasprzak

Yousefff1, 2022-03-19:
Thank you! That did it. It seems the ESP8266 liked the **uint16_t**, even though it was sent as **int** from the Atmega328P-AU.

KrisKasprzak, 2022-03-19:
Yep, glad it was that easy. int's on 32 bit mcu are 4 bytes not 2 like UNOs

closed 2022-03-19 by KrisKasprzak