KrisKasprzak/EBYTE #51: 3 transmissions instead of 1

Yousefff1, 2022-03-29:
Hi,

I have two E220's connected to two Atmega328Ps, they were working just fine, but just recently, instead of sending one transmission every 10 minutes, it sends three at the same time, I've verified this is from the sender, because I have another receiver that also received three  transmissions, this only happened overnight for some reason, I've reuploaded the code, but still, it sometimes randomly sends three transmissions for some reason.

Any ideas?

KrisKasprzak, 2022-03-29:
without any code or wiring, I have no idea.
On 5v0 units (like those arduinos), a pullup between the modules on aux and tx may be help.

Thanks,
Kris ***@***.***

Yousefff1, 2022-03-31:
@KrisKasprzak it's also interesting that sometimes the serial number is received as 256000 not 100, this seems like a norm.

KrisKasprzak, 2022-03-31:
Try sending and receiving the data directly with serial commands

for the receiver:
ESerial.readBytes((uint8_t*)& MyData, (uint8_t) sizeof(MyData));

for the transmitter
ESerial.write((uint8_t*) &MyData, (uint8_t) sizeof(MyData));

Yousefff1, 2022-03-31:
Thank you! I'll try it out.
I'll also change the serial_number from int to uint16_t in the first lines of the sketch.

suiciniv87, 2022-10-16:
> Try sending and receiving the data directly with serial commands
> 
> for the receiver: ESerial.readBytes((uint8_t*)& MyData, (uint8_t) sizeof(MyData));
> 
> for the transmitter ESerial.write((uint8_t*) &MyData, (uint8_t) sizeof(MyData));

I'm trying this, but I'm receiving only unrecognized chars (like bytes) yet.
[image]

Yousefff1, 2022-11-04:
Hi Kris,

Do you have the default configuration values for the E220 433mhz module?

I've been playing around with the configuration sittings, and my E220 modules are no longer receiving or transmitting anything.

It seems the modules stopped transmitting anything after uncommenting Transceiver.SetPullupMode(1);

KrisKasprzak, 2022-11-05:
The default settings can be found on the data sheet.

Thanks.
Kris Kasprzak

closed 2023-02-01 by KrisKasprzak