KrisKasprzak/EBYTE #52: Putting E220 to sleep when not needed.

Yousefff1, 2022-03-31:
Is this a proper way of putting E220 to sleep when not needed?

`digitalWrite(PIN_M0, LOW);
digitalWrite(PIN_M1, HIGH);`

And wake it up using:

`digitalWrite(PIN_M0, LOW);
digitalWrite(PIN_M1, LOW);`

Thank you!

KrisKasprzak, 2022-03-31:
This is not the correct way to activate modes on the ebyte module. When setting a mode, the unit will not immediately enter that mode. The AUX pin will signal when the operation is completed. There is a method on the EBYTE class that does all the heavy lifting

yourobject.SetMode( xxx );

where xxx can be one of these defines

#define MODE_NORMAL 0			// can send and recieve
#define MODE_WAKEUP 1			// sends a preamble to waken receiver
#define MODE_POWERDOWN 2		// can't transmit but receive works only in wake up mode
#define MODE_PROGRAM 3			// for programming

examples of usage
yourobject.SetMode( MODE_POWERDOWN );
yourobject.SetMode( MODE_WAKEUP );
yourobject.SetMode( MODE_NORMAL );

You should be able to send/receive in MODE_WAKEUP

closed 2022-03-31 by KrisKasprzak

Yousefff1, 2022-03-31:
Thank you so much for the quick reply, I honestly was not sure as I couldn't find any example sketches, so I dug into the library and started testing.
My method got the current draw of my entire circuit to less than 1mA, which is perfect!

I'll implement the changes according to your comment, this device will be sealed in a box for years, so removing any bugs now is critical.

Thank you!!