KrisKasprzak/EBYTE #53: Delivery confirmation E220

Yousefff1, 2022-03-31:
Hi @KrisKasprzak,

Is there any way that E220 would know if the transmitted message was delivered? I've been doing some research on this and it seems there is not.

I am currently using a very crude way to ensure delivery:

- Transmitter sends a message, and listens for a reply, entering a while with a loop, it keeps sending the data every 20 seconds and listening.  
- Receiver receives the message and sends the confirmation.
- Once the sender gets the confirmation it breaks the while statement and goes to sleep.

Thank you so much for your help!!

KrisKasprzak, 2022-03-31:
You logic is about the only way to know if a message was received.

KrisKasprzak, 2023-02-01:
There is no way to know automatically. Doing this manually would require sending a confirmation back after receiving a message. This is not technically difficult, on the transmitter side, once a message is sent, start listening for a reply. You should add some type of timer to keep the listen from waiting forever.

closed 2023-02-01 by KrisKasprzak