KrisKasprzak/EBYTE #54: Backwards compatability issues Ebyte E32-900T20D to E32-868T20D

Oldmaker49, 2022-05-13:
This is an issue I have raised with EBBYTE. 
Legacy  E32-868T20D modules (no longer available) have compatibility issues with the replacement E32-900T20D V8.1 modules.
I have been advised by EBYTE that they will only communicate with Air Data Rate of 9600kbps. I have tested this and when the 868 is the Tx and the 900 the Rx it works. But there is some problems when the 900 transmits to the 868. I raised this with EBYTE and they recommend only using E32-900T20D together. 
This is a problem when expanding existing legacy systems since the E32-868T20D is no longer available.

KrisKasprzak, 2023-02-01:
not sure what I can do here. closing.

closed 2023-02-01 by KrisKasprzak

Oldmaker49, 2023-02-01:
Hi Kris,

For information only