KrisKasprzak/EBYTE #57: E32 433T30D transmission not received

abhignay, 2022-10-07:
Hi Kris, firstly thanks for making this library! I'm having some problems with receiving data packets.

I'm using the E32 433T30D 1W module and a Teensy 4.0 with the transmitter and Teensy 3.2 with the receiver. I tried powering the transmitter E32 with a Panasonic NCR18650PF 3c 2900mAh Li-Ion battery, but the E32 was got extremely hot, so I'm currently powering both with the onboard teensy  vregs.

This is the Transmitter code:-

`#include "EBYTE.h"

// connect to any of the Teensy Serial ports
#define ESerial Serial1

#define PIN_M0 4
#define PIN_M1 3
#define PIN_AX 8

// i recommend putting this code in a .h file and including it
// from both the receiver and sender modules

// these are just dummy variables, replace with your own
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};

int Chan;
DATA MyData;

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX);

void setup() {

  Serial.begin(9600);

  while (!Serial) {}

  // start the transceiver serial port--i have yet to get a different
  // baud rate to work--data sheet says to keep on 9600
  ESerial.begin(9600);

  Serial.println("Starting Sender");

  // this init will set the pinModes for you
  Transceiver.init();

  // all these calls are optional but shown to give examples of what you can do

  Serial.println(Transceiver.GetAirDataRate());
  Serial.println(Transceiver.GetChannel());

  Transceiver.SetAddressH(1);
  Transceiver.SetAddressL(0);
  Chan = 4;
  Transceiver.SetChannel(Chan);
  // save the parameters to the unit,
  Transceiver.SaveParameters(PERMANENT);

  // you can print all parameters and is good for debugging
  // if your units will not communicate, print the parameters
  // for both sender and receiver and make sure air rates, channel
  // and address is the same
  Transceiver.PrintParameters();

}

void loop() {

  // measure some data and save to the structure
  MyData.Count++;
  MyData.Bits = analogRead(A0);
  MyData.Volts = MyData.Bits * ( 3.3 / 1024.0 );

  // i highly suggest you send data using structures and not
  // a parsed data--i've always had a hard time getting reliable data using
  // a parsing method
  Transceiver.SendStruct(&MyData, sizeof(MyData));

  
  // You only really need this library to program these EBYTE units. 
  // for writing data structures you can call write directly on the EBYTE Serial object
  // ESerial.write((uint8_t*) &Data, PacketSize );
  
  // let the use know something was sent
  Serial.print("Sending: "); Serial.println(MyData.Count);

  delay(1000);

}`

This is the receiver's code:

`#include "EBYTE.h"

// connect to any of the Teensy Serial ports
#define ESerial Serial1

#define PIN_M0 10
#define PIN_M1 8
#define PIN_AX 12

// i recommend putting this code in a .h file and including it
// from both the receiver and sender modules

// these are just dummy variables, replace with your own
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};

int Chan;
DATA MyData;
unsigned long Last;

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX);

void setup() {

  Serial.begin(9600);

  // wait for the serial to connect
  while (!Serial) {}

  // start the transceiver serial port--i have yet to get a different
  // baud rate to work--data sheet says to keep on 9600

  ESerial.begin(9600);

  Serial.println("Starting Reader");

  // this init will set the pinModes for you
  Transceiver.init();

  // all these calls are optional but shown to give examples of what you can do

  Serial.println(Transceiver.GetAirDataRate());
  Serial.println(Transceiver.GetChannel());

  Transceiver.SetAddressH(1);
  Transceiver.SetAddressL(0);
  Chan = 4;
  Transceiver.SetChannel(Chan);
  //save the parameters to the unit,

  // you can print all parameters and is good for debugging
  // if your units will not communicate, print the parameters
  // for both sender and receiver and make sure air rates, channel
  // and address is the same
  Transceiver.PrintParameters();

}

void loop() {

  // if the transceiver serial is available, proces incoming data
  // you can also use ESerial.available()
  if (Transceiver.available()) {

    // i highly suggest you send data using structures and not
    // a parsed data--i've always had a hard time getting reliable data using
    // a parsing method
    Transceiver.GetStruct(&MyData, sizeof(MyData));
    
    // You only really need this library to program these EBYTE units. 
    // For reading data structures, you can call readBytes directly on the EBYTE Serial object
    // ESerial.readBytes((uint8_t*)& MyData, (uint8_t) sizeof(MyData));

    // dump out what was just received
    Serial.print("Count: "); Serial.println(MyData.Count);
    Serial.print("Bits: "); Serial.println(MyData.Bits);
    Serial.print("Volts: "); Serial.println(MyData.Volts);
    // if you got data, update the checker
    Last = millis();
  }
  else {
    // if the time checker is over some prescribed amount
    // let the user know there is no incoming data
    if ((millis() - Last) > 1000) {
      Serial.println("Searching: ");
      Last = millis();
    }

  }

}`

Here's all the transmitter's parameters:
[image]

The receiver's parameters:
[image]

As you can see in the screenshots above, the transmitter has successfully sent data packets yet the receiver is still searching. Could you help me with this?

Thanks,
moonman

KrisKasprzak, 2022-10-07:
Unplug the 1 watt units from your teensy immediately. These units can easily consume more power that the onboard regulator can supply. I've burned up some NANO's forgetting about power consumptions. I'm surprised the regulator survived this long. 

These units really don't like 5v0 power supply. 3v3 is their recommend input voltage. I hope the voltage / heat issue did not damage the EBYTES

Do you have the antennas connected? The settings look like they match on both units. These should send/receive.

abhignay, 2022-10-08:
Hi Kris, what do you recommend I power the Ebytes with, should I get another one just to be safe?

KrisKasprzak, 2022-10-08:
I power all of mine with DC to DC buck converters there’s a wide variety of them on Amazon some have fixed output voltages some are variable. I’ve used several different types just as long as you’re out putting 3.3 V you should be good and at least one amp. Hard to see if your unit is fried. Do you have antennas installed?

abhignay, 2022-10-08:
Yep, I have antennas installed, What else could be causing the problems? I believe the current and voltage requirements are being fulfilled with my setup

KrisKasprzak, 2022-10-08:
Can you indicate exactly what you’re wearing connections are?

KrisKasprzak, 2022-10-08:
I seriously doubt it will matter between a 3.2 and a 4.0 since they’re both 32 bit but each my pack that struct differently Try sending  just one bite

KrisKasprzak, 2022-10-08:
I’ll knock this up sometime this weekend and see if there are differences between how three point twos and 4.0’s interact

abhignay, 2022-10-08:
The connectors are SMA, here's a picture of them hooked up to the antenna.
[image]

I have a spare Teensy 3.2, I'll try using two 3.2's if I'm still not receiving data packets would it be safe to assume that one of my Ebyte's are fried?

abhignay, 2022-10-09:
Hi Kris. Looks like it's the same result with 2 Teensy 3.2's
[image]

Should I look into buying another Ebyte? On another note would it be OK to power said Ebyte (transmitter) with a 1s 3.6V LiPo?

KrisKasprzak, 2022-10-10:
I think I see what this issue is... I tried your exact code and mine did not work either. I noticed your receiver code does not have the Transceiver.SaveParameters(PERMANENT); Call

I just tried the following (with the Transceiver.SaveParameters(PERMANENT); call)

Teensy LC -> Teensy 4.0
EBYTE E51-TTL-500 (really should not matter).

Mine works, transmits and receives..

Try this...

in your receiver code  just after
//save the parameters to the unit,

add the line

Transceiver.SaveParameters(PERMANENT);

This should get things working.

abhignay, 2022-10-10:
hmm, seems like that didn't fix it, tried it with both these configurations:-

Teensy 3.2 -> Teensy 3.2
Teensy 4.0 -> Teensy 3.2

[image]

One thing I did do is uncomment the defines for the 1W modules in the header file, that didn't help either
[image]

KrisKasprzak, 2022-10-10:
You only need the library to program these units, let's skip the library completely and just use the serial port.

If this does not work, i'd say your units are defective. 

// transmitter
#define ESerial Serial1
//note serial1 is pin 0 and 1 in the MCU, EBYTE Rx connects to pin 1, EBYTE Tx connects to pin 0
// you could try Serial2 or Serial3 Teensy 4.0 pin 7, 8 are Serial2, Teensy 3.2 pin 7, 8 are Serial3
#define PIN_M0 4
#define PIN_M1 3
#define PIN_AX 8

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&ESerial, PIN_M0, PIN_M1, PIN_AX);

void setup() {
 pinMode(PIN_MO, OUTPUT);
 pinModePIN_M1 OUTPUT);

digitalWrite(PIN_MO, LOW);
digitalWrite(PIN_M1, LOW);

  Serial.begin(9600);

  while (!Serial) {}

  ESerial.begin(9600);

  Serial.println("Starting Sender");

}

void loop() {

  ESerial.print(9);

  Serial.print("Sending");

  delay(1000);

}

////////////////////////////////////////
// receiver

#define ESerial Serial1

//note serial1 is pin 0 and 1 in the MCU, EBYTE Rx connects to pin 1, EBYTE Tx connects to pin 0
// you could try Serial2 or Serial3 Teensy 4.0 pin 7, 8 are Serial2, Teensy 3.2 pin 7, 8 are Serial3
#define PIN_M0 10
#define PIN_M1 8
#define PIN_AX 12

void setup() {
 pinMode(PIN_MO, OUTPUT);
 pinModePIN_M1 OUTPUT);

digitalWrite(PIN_MO, LOW);
digitalWrite(PIN_M1, LOW);
  Serial.begin(9600);

  while (!Serial) {}

  ESerial.begin(9600);

  Serial.println("Starting Reader");

}

void loop() {

  if ESerial.available()) {

	Serial.println(ESerial.read();

  }
  else {
    // if the time checker is over some prescribed amount
    // let the user know there is no incoming data
    if ((millis() - Last) > 1000) {
      Serial.println("Searching: ");
      Last = millis();
    }

  }

}

abhignay, 2022-10-10:
Looks like the my transmitter is defective, also would it be ok to power said Ebyte (transmitter) with a 1s 3.6V LiPo? 
[image]

KrisKasprzak, 2022-10-10:
I would check the data sheet for an acceptable range of supply voltages

KrisKasprzak, 2022-10-10:
one last thing, put a delay(10) in the receiver loop just before the available() call.

I noticed my Teensy 4.0 sometimes can't ready the serial port from my GPS sensor,

abhignay, 2022-10-11:
Alright Kris! I'll try that, would you mind if I close this issue once I get the new Ebyte (should be in 2 days) and test it?

closed 2022-10-11 by KrisKasprzak

abhignay, 2022-10-12:
Hi Kris, just wanted to end this issue with this comment. I managed to get everything working with a new Ebyte! 
Thanks for all your help and making such an awesome library :)

[image]

KrisKasprzak, 2022-10-12:
That is great news. I find these units very reliable even if loss of power, they start right up and continue working. I have these in  a home automation setting (remote garage door and outside temp monitoring) and have been working for about 2 years with no issues.

If you find yourself wanting to communicate between 2 different MCU (a Teensy and NANO), the structure will probably pack differently and become garbled up on the receive, __attribute (packed)___ may help, or use the EasyTransfer library.

abhignay, 2022-10-12:
Ok Kris, will keep that in mind!

abhignay, 2022-10-12:
Hi Kris, one last thing, the datasheet recommends using this module with 5V, would it be ok to power it with 5V @ 3A (using an [AP63300](https://www.diodes.com/assets/Datasheets/AP63300-AP63301.pdf) to stepdown 7.4V to 5V, 7.4V comes from a 2s Li-Po). 

[image]

KrisKasprzak, 2022-10-12:
If that’s what the data sheet says… I’ve always powered mine with 3.3 they seem to get warm at 5 V. However before you connect your RX and TX lines I would measure the output voltage and make sure there’s still 3.3 if you power from 5 volts TEensy 4 are not 5 V tolerant. I’m not sure about that specific buck converter but that’s all I use.

Thanks.
Kris Kasprzak