KrisKasprzak/EBYTE #58: Can I make the library non-blocking?

abhignay, 2022-10-18:
Hi Kris, you probably know that the library freezes all code when a packet of data is sent. could you help me with making this library "non blocking"

abhignay, 2022-10-29:
Hi Kris, sorry to bother you again but could you help me with my issue?

closed 2022-11-07 by KrisKasprzak

KrisKasprzak, 2022-12-11:
If you are sending a struct, just send directly and not use the EBYTE::SendStruct

MyData is your struct

ESerial.write((uint8_t*) &MyData, (uint8_t) sizeof(MyData) );

This write to the Serial buffer and returns control back to your code.

Onus will be on you to know when the transmission is complete, but my experience has been these send very quickly

KrisKasprzak, 2022-12-11:
or comment out the CompleteTask line

bool EBYTE::SendStruct(const void *TheStructure, uint16_t size_) {
		_buf = _s->write((uint8_t *) TheStructure, size_);
		CompleteTask(1000); // comment out this to return control immediately back to your code
		return (_buf == size_);
}

abhignay, 2022-12-13:
Hi Kris, tried both of your suggestions, but it seems like the issue is still persistent. The E32 still delays all the code when it sends a struct.

abhignay, 2022-12-15:
Hi Kris, it seems like sending multiple structs a couple of milliseconds after each other (instead of a big struct with all the data I want to send) and adding a simple `ESerial.flush();` fixed things on the Transmitter side. (I am using `ESerial.write();` to send the struct)

However it looks like my data is corrupted when received (with both an `ESerial.readBytes((uint8_t*)&Mydata, sizeof(Mydata));` and the regular `Transceiver.GetStruct(&MyData, sizeof(MyData));`)

This is the data being received
[image]

And this is the data being sent
[image]

abhignay, 2022-12-20:
Hi Kris, 

After a little more debugging it appears that commenting out the `CompleteTask(1000);` (when using the `Transceiver.GetStruct` and `Transceiver.SendStruct`, data is corrupted any which way when using `ESerial.write`) is making the data corrupted, I get normal data when the `CompleteTask(1000);` is not commented out. Could you suggest a fix for this?

Thanks,
moonman

KrisKasprzak, 2022-12-20:
According to the datasheet you must listen to the AUX pin to know when a transmission is complete. That is what complete task is doing. Commenting it out and skipping that step may be what is causing corrupt data. It's really impossible to debug your issue with the limited information provided. I'm not sure why you need non-blocking--you can't keep sending data until the buffer has been sent. It sounds like you want to send data to the EBYTE, return control immediately, then send more data without waiting for the transmission to happen. Again, I'm really not understanding what you are trying to do.

Note that this library is really only needed if settings need to be changed. I can only suggest you write directly to the EBYTE serial port and listen directly to the EBYTE serial port, use the lib if you need to change a setting during program execution.

I don't see how the transmission process can be made non-blocking.