KrisKasprzak/EBYTE #63: Sending Array bytes

PadmaIoT, 2023-06-05:
Hello, 
Thanks for this library.
I want to send a byte array of 42 members in it. without using struct how can I send and receive the byte array? I am trying to simultaneously send & receive to & from 3 E32 device units. Is it possible? If possible then how? Can you guide me? Please I will be obliged for your help.

KrisKasprzak, 2023-06-26:
I've tried many methods in sending data w/o a struct and all methods ended up being unreliable. Not sure why you can't use a struct but I've found it to be the most reliable and easy to implement.

something like this

struct DATA {
  byte data[42];
};

DATA Kris;

void setup() {
  Serial.begin(9600);
  while (!Serial){}
  Kris.data[0] = 4;
  Kris.data[1] = 6;
}

void loop() {
// here you send or listen to the EBYTE serial port
Serial.println(Kris.data[0]);
}

Regarding sending / receive multiple, it's not clear what you are trying to do. 1) send 1 to many 2) send many to one 3) other. I have a system where one sends to many, but I'm not sure if many can send to one.

Note these units only work in half duplex.

I'll need a bit more info to assist further.

PadmaIoT, 2023-06-26:
Hello,
Thanks for your kind help.
My system

closed 2023-06-26 by PadmaIoT