KrisKasprzak/EBYTE #64: E32 433T30D AUX pin constant low signal

Fompi06, 2023-06-26:
Hello, I have two LoRa E32 433T30D. I connect to a computer via USB-TTL E15-USB-T2:
1. One module works successfully, when you enter something into Serial, AUX-LED blinks for a few milliseconds, the program "Lora E32 Setting" connect with LoRa
2. The second module also worked successfully before, it was connected to Arduino Mega via Serial1. Now he does not want to accept and give the data. When connected to USB-TTL, the AUX-LED is always on. It does not respond to Serial in any way. The program "Lora E32 Setting" is not connect.

What could it be because of? Are there any ways to solve this problem?

P.S. I used translator, sorry for errors in text.

BJPatriotRacing, 2023-06-26:
Arduino Mega will use a 5V0 voltage level for Tx. According to the datasheet (Page 5) you are only supposed to supply 3V3 to signal pins on the EBYTE. Could it be damaged? Possibly. Try using voltage dividers on the EBYTE Rx pin to drop the voltage to ~3V3.

Fompi06, 2023-06-26:
> 

Yes, that's right, we use a board that has a 3.3 V voltage divider. So there shouldn't be any problems with the power supply.

closed 2024-03-31 by Fompi06