KrisKasprzak/EBYTE #65: Increasing transmission rate of Ebyte E34 2G4D20D

gbaranedavid, 2023-06-26:
``Hello Kris, thanks for the ebyte library provided and the video intro provided at https://www.youtube.com/watch?v=hMjArKGucFA

So I decided to modify the library, specifically, increasing the transmission rate per second a bit to suit my needs. At first, I observed that t when I call the SendStruct(const void *TheStructure, uint16_t size_) in loop() method, it defaults to 1 transmission per second and I wanted it faster, lets say up to 12 transmisson per second.  I didn't know how to go about it so I studied the Ebyte.cpp library and adjusted the line of code below: 

`bool EBYTE::GetStruct(const void *TheStructure, uint16_t size_) {
	
	_buf = _s->readBytes((uint8_t *) TheStructure, size_);

	CompleteTask(100);

	return (_buf == size_);
}`

`bool EBYTE::SendStruct(const void *TheStructure, uint16_t size_) {

		_buf = _s->write((uint8_t *) TheStructure, size_);
		
		CompleteTask(100);
		
		return (_buf == size_);

}`

`void EBYTE::CompleteTask(unsigned long timeout) {

	unsigned long t = millis();

	// make darn sure millis() is not about to reach max data type limit and start over
	if (((unsigned long) (t + timeout)) == 0){
		t = 0;
	}

	// if AUX pin was supplied and look for HIGH state
	// note you can omit using AUX if no pins are available, but you will have to use delay() to let module finish
	
	// per data sheet control after aux goes high is 2ms so delay for at least that long
	// some MCU are slow so give 50 ms
	
	if (_AUX != -1) {
		
		while (digitalRead(_AUX) == LOW) {
			delay(PIN_RECOVER);
			if ((millis() - t) > timeout){
				break;
			}
		}
	}
	else {
		// if you can't use aux pin, use 4K7 pullup with Arduino
		// you may need to adjust this value if transmissions fail
		delay(100);

	}

	// delay(PIN_RECOVER);
}`

I observed that transmission rate increased, so I uploaded same code on a second device, but observed that I wasn't receiving what works is Tx but no trace of Rx.

Below is the sample source code I am using for Tx and Rx: 

`#include "EBYTE.h"

#define PIN_RX 16   // Serial2 RX (connect this to the EBYTE Tx pin)
#define PIN_TX 17   // Serial2 TX pin (connect this to the EBYTE Rx pin)

#define PIN_M0 4    // D4 on the board (possibly pin 24)
#define PIN_M1 22   // D2 on the board (possibly called pin 22)
#define PIN_AX 21   // D15 on the board (possibly called pin 21)

// i recommend putting this code in a .h file and including it
// from both the receiver and sender modules
struct DATA {
  unsigned long Count;
  int Bits;
  float Volts;
  float Amps;

};

// these are just dummy variables, replace with your own
int Chan;
DATA MyData;
DATA MyDataTx;
unsigned long Last;

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&Serial2, PIN_M0, PIN_M1, PIN_AX);

TaskHandle_t Task1;

// LED pins
 int counter1 = 0;
 int counter2 = 0;

void setup() {
  Serial.begin(9600);

  Serial2.begin(9600);

  Serial.println("Starting Reader");

  // this init will set the pinModes for you
  Serial.println(Transceiver.init());

  // all these calls are optional but shown to give examples of what you can do

Serial.print("Data rate: ");
  Serial.println(Transceiver.GetAirDataRate());
  Serial.print("Channel: ");
  Serial.println(Transceiver.GetChannel());
  Serial.print("Baud rate: ");
  Serial.println(Transceiver.GetUARTBaudRate());
  Serial.print("Transmission mode: ");
  Serial.println(Transceiver.GetTransmissionMode());
  // and address is the same
  Transceiver.PrintParameters();

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    80000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    tskIDLE_PRIORITY,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 
}
void loop() {
  // Main core used for sending
  Serial.print("Tx Task running on main core ");
    MyDataTx.Count=333;
    MyDataTx.Bits = analogRead(A0);
    MyDataTx.Volts = 9.99;
  
    Transceiver.SendStruct(&MyDataTx, sizeof(MyDataTx));
    
    Serial.print("Sending: "); Serial.println(MyDataTx.Count);
   // vTaskDelay(10/portTICK_PERIOD_MS);
}
//Task1code:Receiver task
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
  
    if (Serial2.available()) {
  
      Transceiver.GetStruct(&MyData, sizeof(MyData));
      // dump out what was just received
      Serial.print("Count: "); Serial.println(MyData.Count);
      Serial.print("Bits: "); Serial.println(MyData.Bits);
      Serial.print("Volts: "); Serial.println(MyData.Volts);
      // if you got data, update the checker
      Last = millis();
  
    }
    else {
      // if the time checker is over some prescribed amount
      // let the user know there is no incoming data
      if ((millis() - Last) > 1000) {
        Serial.println("Searching: ");
        Last = millis();
      }
  
    }
  } 
}`

My goal is to increase transmission rate from one transmission per second to 12 or more transmission per second.
Any push in the right direction will be much appreciated.

Thanks

gbaranedavid, 2023-07-05:
@KrisKasprzak Any info on this would be much appreciated, I would like to have your opinion on this before moving forward

KrisKasprzak, 2023-07-05:
If you increase the USB baud rate and air data rate, you will get faster data rates. I'm not sure how much data you want to transmit. More importantly, I'm not sure over what distance--that too will make a difference in how fast you want to move data.

The air data rate I believe can run up to 57.6 which will help. I hate to say it but the only suggestion I have is to get a few units and test it out.

gbaranedavid, 2023-07-05:
Thank for the response Kris, will try that out.
I will also restore  CompleteTask back to 1000ms
hope it won't slow it down?

KrisKasprzak, 2023-07-05:
As you probably found the argument for CompleteTask is a timeout to prevent an infinite loop. The time for aux pin to return high should be 2 ms according to the datasheet. Depending on the MCU, this can take longer. My experience is using a Cortex processor (Teensy MCU), time to see aux pin go high is < 10 ms.  

I have a provision for projects that run out of pins, users can omit the aux pin and hope it goes high in 1 sec (see line 224 in the .cpp file). 

not sure this helps, but keep us posted.

gbaranedavid, 2023-07-06:
Ok Kris! we will work on this and update you on how it turns out.

Thanks Kris!

gbaranedavid, 2023-08-05:
Hello @KrisKasprzak, I  implemented fix. however,  transmission rate spikes up and after a moment, returns to transmitting per second again.... any way we can make the fast transmission consistent.