KrisKasprzak/EBYTE #66: Sending data both ways, how to use Half Duplex

embeddedDevEnthusisast, 2023-07-28:
Hello! I have been using this library for a while for my embedded systems projects. I am wondering how to use Half-Duplex data communication with this library, that is sending data both ways but at different times between two transciever modules.

Thanks

embeddedDevEnthusisast, 2023-08-14:
Hi, can anyone help me please?

Il-Minguz, 2024-04-15:
> Hello! I have been using this library for a while for my embedded systems projects. I am wondering how to use Half-Duplex data communication with this library, which is sending data both ways but at different times between two transciever modules.
> 
> Thanks

So, technically it is pretty simple. You have to call in the two modules this function in the void loop:if (Serial2.available()) {.
if you need to send something just make the Arduino some stuff and then use the command:Transceiver.SendStruct(&MyData, sizeof(MyData));
Here's an example:
void loop() {
  if (Serial2.available()) {
    Transceiver.GetStruct(&MyData, sizeof(MyData));
    Serial.print("aperto: ");
    Serial.println(MyData.aperto);
 if (MyData.aperto) {
    passwordGen(5);
    Transceiver.SendStruct(&MyData, sizeof(MyData));
  }
  }
}

KrisKasprzak, 2024-04-16:
I thought I responded--I do this all the time

Have all devices in "listen mode" meaning in a loop keep looking for incoming data, in that loop, send when needed, since others are always listening, they will get the message

something like this....

loop(){

// measure data
if (enoughdata){
// build struct
needToSend = true;
}

If (needToSend){
needToSend = false;
// send data as shown in examples
}
delay(50); // give time for EBYTE to do it's thing

if (EBYTESeria.available()) {
// read data as shown in examples
// process received data
// this could also be a function to process data and send right back out (see measure data section)
}

closed 2024-04-16 by KrisKasprzak