KrisKasprzak/EBYTE #67: Display, Transmit and Receive RSSI values on Ebyte E49-400T30D

ahwahtai, 2023-12-08:
Hi Kris, thanks for the providing the ebyte library and also for the introduction on https://www.youtube.com/watch?v=hMjArKGucFA

I am using two Arduino Uno's and two E49-400T30D units. I have been trying to study the code in the libraries and also the datasheet but am not able to understand how to display, transmit, and/or receive the RSSI values after setting the M0 to "LOW" and M1 to "HIGH" on either unit

[E49-400T30D_Usermanual_v1.2EN.pdf](https://github.com/KrisKasprzak/EBYTE/files/13611060/E49-400T30D_Usermanual_v1.2EN.pdf)

Would you be willing to help me with this? Thanks!

KrisKasprzak, 2023-12-08:
The EBYTE modules you are using do not support RSSI, the company does have some that support RSSI. This library was written before RSSI modules were available and does not support RSSI features.

closed 2023-12-08 by KrisKasprzak

ahwahtai, 2023-12-13:
Thank you for responding so quickly, Kris. It is much appreciated!

Hmmm ... that is very strange! I am reading in the datasheet (section 6: Operating mode) and 6.3 that the unit will output a strength value (relative value) through the serial port every 100ms if the RSSI mode is enabled (mode 1). I hope I am not reading it wrongly?

KrisKasprzak, 2023-12-13:
ahhh I was reading the wrong section. This lib does not support reading RSSI but you can read it directly

You would simply drive M0 HIGH, M1 LOW, then read the serial buffer

something like this untested code
digitalWrite(M0_PIN, HIGH);
digitalWrite(M1_PIN, LOW);

while (1) { // don't do this, need a better way to prevent runaway loops
   if (Serial.available()){
      uint8_t data= Serial.read();
      int rssi = -(256 - data);
     break;
   }
}