KrisKasprzak/EBYTE #70: String

JayZeeM, 2024-02-09:
Hello I tried using the send and receive examples of ESP32 and changing the structure message with

String message;

and edited the other parts as well but on the receiver side it is not showing the message I inputed which is "Hello"

KrisKasprzak, 2024-02-09:
I recommend changing message to a char of fixed length. I'm not sure you can use strings in a struct as the length is unknown.

closed 2024-02-09 by KrisKasprzak

JayZeeM, 2024-02-09:
Do you have an example code for a char with fixed length?

KrisKasprzak, 2024-02-09:
Not tested but it at least compiles

// create struct for sender and receiver
struct Transceiver {
	char name[10];
} ;

// populate data on sender side
Transceiver  Data;
strcpy(Data.name, "kris"); // keep length under 10

// on receiver after you got data
Serial.println(Data.name);

KrisKasprzak, 2024-02-09:
PS... chars are very different and to be honest can be pain full... Pointers, strcat, strcpy and other non-intuitive functions to perform very basic string operations. But... chars are native to C and Strings are not. In my early day of the Arduino world 've spent many months chasing random crashes and memory  issues created from string fragmentation.

JayZeeM, 2024-02-10:
`/*

  This example shows how to connect to an EBYTE transceiver
  using an ESP32

/*

  This example shows how to connect to an EBYTE transceiver
  using an ESP32

  This code for for the sender

  ESP32 won't allow SoftwareSerial (at least I can't get that lib to work
  so you will just hardwire your EBYTE directly to the Serial2 port

*/

#include "EBYTE.h"

/*
WARNING: IF USING AN ESP32
DO NOT USE THE PIN NUMBERS PRINTED ON THE BOARD
YOU MUST USE THE ACTUAL GPIO NUMBER
*/
#define PIN_RX 1   // Serial2 RX (connect this to the EBYTE Tx pin)
#define PIN_TX 3     // Serial2 TX pin (connect this to the EBYTE Rx pin)

#define PIN_M0 18    // D4 on the board (possibly pin 24)
#define PIN_M1 19   // D2 on the board (possibly called pin 22)
#define PIN_AX 21   // D15 on the board (possibly called pin 21)

// i recommend putting this code in a .h file and including it
// from both the receiver and sender modules
struct DATA {
  char message[10];

};

// these are just dummy variables, replace with your own
int Chan;
DATA MyData;
unsigned long Last;

// create the transceiver object, passing in the serial and pins
EBYTE Transceiver(&Serial2, PIN_M0, PIN_M1, PIN_AX);

void setup() {

  Serial.begin(9600);

  Serial2.begin(9600);
  Serial.println("Starting Reader");

  // this init will set the pinModes for you
  Transceiver.init();

  // all these calls are optional but shown to give examples of what you can do

  // Serial.println(Transceiver.GetAirDataRate());
  // Serial.println(Transceiver.GetChannel());
  Transceiver.SetAddressH(1);
  Transceiver.SetAddressL(1);
  Chan = 18 ;
  Transceiver.SetChannel(Chan);
  // save the parameters to the unit,
  // Transceiver.SetPullupMode(1);
  // Transceiver.SaveParameters(PERMANENT);

  // you can print all parameters and is good for debugging
  // if your units will not communicate, print the parameters
  // for both sender and receiver and make sure air rates, channel
  // and address is the same
  Transceiver.PrintParameters();

}

void loop() {

  // if the transceiver serial is available, proces incoming data
  // you can also use Transceiver.available()

  if (Serial2.available()) {

    // i highly suggest you send data using structures and not
    // a parsed data--i've always had a hard time getting reliable data using
    // a parsing method

    Transceiver.GetStruct(&MyData, sizeof(MyData));

    // note, you only really need this library to program these EBYTE units
    // you can call readBytes directly on the EBYTE Serial object
    // Serial2.readBytes((uint8_t*)& MyData, (uint8_t) sizeof(MyData));

    // dump out what was just received
    Serial.print("message"); Serial.println(MyData.message);
    // if you got data, update the checker
    delay(2000);

  }
  else {
    // if the time checker is over some prescribed amount
    // let the user know there is no incoming data
    if ((millis() - Last) > 1000) {
      Serial.println("Searching: ");
      Last = millis();
    }

  }
}
`

this is my receiver code and my Serial monitor is only printing this Model no.: 0
Version  : 0
Features : 0
 
Mode (HEX/DEC/BIN): 0/0/0
AddH (HEX/DEC/BIN): 1/1/1
AddL (HEX/DEC/BIN): 1/1/1
Sped (HEX/DEC/BIN): 0/0/0
Chan (HEX/DEC/BIN): 12/18/10010
Optn (HEX/DEC/BIN): 0/0/0
Addr (HEX/DEC/BIN): 101/257/100000001
 
SpeedParityBit (HEX/DEC/BIN)    : 0/0/0
SpeedUARTDataRate (HEX/DEC/BIN) : 0/0/0
SpeedAirDataRate (HEX/DEC/BIN)  : 0/0/0
OptionTrans (HEX/DEC/BIN)       : 0/0/0
OptionPullup (HEX/DEC/BIN)      : 0/0/0
OptionWakeup (HEX/DEC/BIN)      : 0/0/0
OptionFEC (HEX/DEC/BIN)         : 0/0/0
OptionPower (HEX/DEC/BIN)       : 0/0/0
----------------------------------------
message
message
message
message
message
message
message

I am using the same channel and address for both LoRa

KrisKasprzak, 2024-02-10:
I use this lib and ESP's all the time and have been successful. all zeros generally indicates wiring where the serial lines are not corrected. Try increasing the pin recovery

in the .h change the 50 to say 100

#define PIN_RECOVER 50 

just for fun call these lines after Transceiver.init()

delay(1000);
Transceiver.PrintParameters();

If you are using the above code, settings are never getting saved (temporarily or permanently). 

// Transceiver.SaveParameters(PERMANENT);

JayZeeM, 2024-02-10:
Is there any soc med that I can talk with you privately sir?

KrisKasprzak, 2024-02-13:
I just tried the example code on and ESP32 (dev kit) and with 3 different EBYTE modules. All fired right up and output the model and correct parameters.

1. What EBYTE modules are you using
2. how are they powered
3. are you sure wiring is correct
4. any other details to help me help you

JayZeeM, 2024-02-13:
E22-900t22d
I powered it only through my laptop and connected the pins to their respective pins. Like AUX M0 M1 to their pins in esp32 and im using TX0 and RX0 because my ESP32 doesn't have TX2 and RX2. Do I have to connect 3.3V on the RX and TX of E22?

JayZeeM, 2024-02-13:
I also use E32 433T20D i have 1 set of each EByte module

KrisKasprzak, 2024-02-13:
that would be the problem can't use serial 0 (or 1) as they are tied to hardware internally. ESP32 has serial2 available on the hardware pins I call out. If you can;t use those pins, you will have to use softwareserial.h and chose different ones 

I have 3 UARTs in one project and I had to use sofwareserial.h

like this....

SoftwareSerial Serial_A(33, 32); // ESP RX, TX
SoftwareSerial Serial_B(21, 22);// ESP RX, TX

EBYTE Transceiver(&Serial_A, PIN_M0, PIN_M1, PIN_AX);

JayZeeM, 2024-02-14:
But I cant use SoftwareSerial.h with ESP32 do you have any sample code?