KrisKasprzak/EBYTE #75: Library uploading not working

ashtonella, 2024-03-09:
hello Kris,

I like your library for Ebyte. simple and easy to follow. 

I tried to uploaded on Arduino custom library but it is not uploading.  is says the library.properties file not found. I also didnt find the .json file attached with your library. is there anything I need to do to debug this on my side?

thx

KrisKasprzak, 2024-03-10:
I’ll need more information than this. What MCU are you using? Are you using the Arduino ide. Exactly. What is the error message. How did you install the library. This is a very odd error.

ashtonella, 2024-03-12:
Hi thanks for replying: 
I did a lot of trouble shooting with the issues I had and I was finally able to get it to work. 
1. I was using the online Arduino create tool to upload the library, that one did not upload automatically because of the .json files and property files, i was eventually able to upload the EBYTE.cpp and EBYTE .h manually on the online IDE.  
2. the library wouldn't compile that was because of the <SerialSoftware> was missing delay.h file, but I was able to disable the two libraries and their members to get the EBYTE to compile and it did. 
3. I eventually downloaded the arduino IDE and uploaded your library and it worked just fine no issues at all. 
4. I used Aduino MMRK 1010 : did not upload the send /receive examples. 
5. I used WAMOS: didn't work. 
6. I used 2 deferent ESP826 MCU : didnt work. 
7. I finally used Arduino Uno and it worked no issues. 
so that was a success. really appreciate the Library. great work!

ashtonella, 2024-05-08:
Hello Kris; 
I was wondering if your library has a method that deals with point-to-point fixed communication for Ebyte modules? 
I also tried to send a String in the Struct, but I received corrupt data. I increased the air rate with no luck still. any ideals how to deal with these issues? 
thanks much in advance. 

Ashton

KrisKasprzak, 2024-05-08:
This library is really for programming these modules only while there is a method to send and receive data. It does nothing more than mimic read Byrd and write byte. You cannot use strings inside of a struct They are an undefined length use char’s

closed 2024-05-08 by KrisKasprzak

ashtonella, 2024-05-09:
Wonderful!will do. Thank you very much. Sent from my iPhone